RespiraWorks/Ventilator #10: Add high/low pressure, high low volume, and patient disconnect alarms

rwilbur, 2020-04-09:
# Requirement
If multiple alarms simultaneously present and display has one line of text for alarms, display line will toggle through all alarms

- **Patient Disconnect**
  - Condition: pressure does not rise above 3 cmH2O during inspiration
  - Audible alarm and UI text “LOW AIRWAY PRESSURE”
- **High inspiratory airway pressure**
  - Condition: airway pressure above alarm setting during inspiration
  - Adjustable range: 15- 60 cmH2O, increments of 5 cmH2O
  - Default setting: 35 cmH2O
  - Audible alarm and UI text “OVER-PRESSURE ALARM”
  - Controller feedback:
     - Inspiratory solenoid valve closes to stop inspiratory flow
     - Exhalation valve opens (if controlled by solenoid)
- **TV High/Low**
  - Adjustable range: 200-1000 ml, increments of 50 ml
  - Default low alarm setting: 200 ml; Default high alarm setting: 1000 ml
  - Audible alarm and UI text: “TV HIGH” 
  - Condition: 
    - measured TV exceeds high TV alarm setting
    - “SET TV NOT DELIVERED” Condition: measured TV less than low TV alarm setting
    - “LOW VT, IT SHORTENED” Condition: Inspiration Time (IT) is shortened by reaching high inspiratory airway pressure limit and measured TV less than low TV alarm setting

martukas, 2020-04-13:
Also, I think it's worthy of discussion whether some of this bounds logic ought to happen on the controller side, so then some of those tickets might be addressed to that component.

martukas, 2020-05-03:
Blocked by #169

martukas, 2021-01-18:
This needs to be cross referenced with issues and fleshed out in separate tickets. Someone can pick up this task.