RespiraWorks/Ventilator #12: Add gas pipeline failure alarm

rwilbur, 2020-04-09:
# Enhancement
- **Gas pipeline failure (Optional)**
  - Condition: no supply pressure detected
  - Audible alarm and UI text “GAS SUPPLY FAILURE”

tapemaster, 2020-05-27:
Slack thread related to this: https://respiraworks.slack.com/archives/C0100AC23MJ/p1590544077112700

dtgerson, 2020-05-30:
For O2 low pressure or O2 disconnects:

Current idea: if we are maxing out the PSOL (commanding it to its highest state) but not able to control patient pressure (can't hit pressure target) or achieve the FiO2 level within the desired tolerance, then we know something is wrong with the oxygen (either disconnect or low pressure). So look for sustained deviations from the control set points (both the intra-cycle and the inter-cycles ones) Note: in parallel, we want to supplement with air - but we should still trigger an oxygen alarm.

For problems with the blower (not a gas supply but related):

Similar: if maxing out the air inhale valve but can't hit the target pressures, then something is wrong

tapemaster, 2020-05-31:
Another discussion on slack here: https://respiraworks.slack.com/archives/C013KQQ18T1/p1590870727028600

jkff, 2020-05-31:
I think there's been some indication that this may be overlapping with https://github.com/RespiraWorks/VentilatorSoftware/issues/368 - Like, having a "low pressure" alarm can be enough to detect disconnection?

dtgerson, 2020-05-31:
I think there are related but slightly different things.

If you have a low FiO2 and you can't hit pressure - then it means something is going wrong, but it is not an issue with the O2 system.

If you have a high FiO2 and you can't hit pressure, but if you compensate with air and are able to hit pressure, then it says that there is something specifically wrong with the O2 system (likely the supply - though could be a failure of the vent/valve).

Unclear to me if we actually want to have two separate alarms for this or just have an alarm of "can't hit pressure" and then let the doctor figure out what is causing it. If we are going to build in logic that compensates with air if we can't hit the pressure at the desired FiO2, then it makes sense to have two alarms (one in general, one for oxygen specifically) - but I don't know if we've decided to add that complexity yet.

Does that make sense?