RespiraWorks/Ventilator #15: PID variables in EEPROM

neelfirst, 2020-04-09:
Copied thread from inceptionev/FixMoreLungs

@rwilbur We will need the variables to be stored in EEPROM so that in case of a reboot we still have the correct values for the variables

@lee-matthews Indeed, this functionality shall be provided by the eeprom module in the software.

martukas, 2020-04-13:
Is this deserving to be in alpha? This is a safety feature in case of reboot, but is it crucial for functioning prototype? Same applies to #56 and #58

lee-matthews, 2020-04-16:
I think we can move this to beta. We can just store these values as constants in flash for the time being.

jlebar, 2020-04-25:
Dupe of https://github.com/RespiraWorks/VentilatorSoftware/issues/56

closed 2020-04-25 by jlebar