RespiraWorks/Ventilator #17: Change PID #defines to variables

neelfirst, 2020-04-09:
Copied thread from inceptionev/FixMoreLungs:

@rwilbur We will need to change the PID #defines to variables so they can be changed via the communications to the raspberry pi

@lee-matthews Indeed. Was planning on making possible to set those too. Maybe create some kind of "engineering/debug" mode on the interface module so that we can play with these values directly that's protected from a typical user.

@rwilbur That makes sense. I am pretty sure that all these will need to be set in the UI but open to a user since nurses and doctors will need to be changing all of these based on the patient and their progress.

#define INSPIRE_TIME 1600
#define INSPIRE_RATE 1
#define PIP 142
#define INSPIRE_DWELL 800
#define INSPIRE_DWELL_PRESSURE 140
#define EXPIRE_TIME 1000
#define EXPIRE_RATE 1
#define PEEP 130
#define EXPIRE_DWELL 600

Perhaps with these we won't have these values being set directly but have it be a certain unit on the display and it gets converted to whatever these units are in the command.

@lee-matthews I was thinking of an engineering mode for the parameters for the PID for example, I don't think the medical personnel should have access to these. Same for any potential calibration parameters.

jlebar, 2020-04-15:
I've looked more closely at this.  There's a lot more to making this module work than just replacing the `#define`'ed variables with `parameter_getFoo()`.

 - Change the module to be time-based, rather than based on the number of times we call it from the main loop (which has no bearing on time).
 - Add separate inhale/exhale-dwell parameters (right now parameters and comms just have a single "dwell" param).
 - Some sort of testing.  I noticed upon code inspection that `cyclecounter` is not a static variable, so doesn't actually count anything between calls.

I also think that we should get rid of most or all of our `float` variables, eventually up through the comms layer.  There's no need to be doing floating-point math, which is expensive, error-prone, and I think unnecessary here.  AIUI our current microcontroller doesn't even support FP; maybe the only reason this compiles is that we're targeting a different chip in our `platformio.ini`?

lee-matthews, 2020-04-16:
Indeed, this needs a lot of work.
As a start I was thinking of changing the name from pid.cpp to something like controller.cpp
Then the controller module would act as an abstraction layer to the two medical ventilation mode modules, PRVC and  ACV. Each one of these would have its own FSM to control the breathing cycle. What do you think?

lee-matthews, 2020-04-16:
I've started layout out my thoughts in this branch : lee-matthews/issue103_controllerModule

Please let me know what you think.

jlebar, 2020-04-16:
> As a start I was thinking of changing the name from pid.cpp to something like controller.cpp
Then the controller module would act as an abstraction layer to the two medical ventilation mode modules, PRVC and ACV. Each one of these would have its own FSM to control the breathing cycle. What do you think?

That sounds smart to me!  My only comment ("two hard problems in computer science") is that maybe we shouldn't use the name "controller", because we're already using that word to refer to the microcontroller chip and code.  blower_controller?  Or just "blower"?  I'm really not sure what blower.cpp is supposed to do.

> I've started layout out my thoughts in this branch : lee-matthews/issue103_controllerModule

What is the base rev for the branch?  I went here and compared it to master, but I think that's (a lot) more changes than are relevant.

https://github.com/RespiraWorks/VentilatorSoftware/compare/lee-matthews/issue103_controllerModule

lee-matthews, 2020-04-16:
Yeah, that's more than likely the case as I screwed up and merged master
into issue 77 instead of the other way around. So I started the branch
issue103 based on branch issue77 so it's understandable there are more
changes in there than necessary compared to master. I'm still getting used
to github.

Yeah, I hear you on the name, I'll think of something else.

Miceuz, 2020-04-25:
One short note - PID loop has to run in a strictly periodic fashion for it to work properly. Now it's called as fast as possible. That will break as soon as we add new code to the main loop - it will de-tune PID loop, particularly Ki and Kd variables.

For PID to run properly it has to be triggered from timer interrupt.

Also in pid.cpp we have two things intermingled - PID loop driving the blower and state machine driving the inspiration/expiration. I think we should split that in two separate files. PID has to run from timer interrupt and inspire/expire fsm can be run from main loop.

jlebar, 2020-04-25:
Ugh, thanks, Albertas.

> That will break as soon as we add new code to the main loop - it will de-tune PID loop, particularly Ki and Kd variables.

And moreover it'll break because the time to run one iteration of our main loop is not fixed, right?

Miceuz, 2020-04-25:
I don't speak phase margins, but as you go from continuous domain to discrete you are fine as long as your time quantification is constant. 

Integral and differential constants are time-dependent they are constant as long as time is constant. If we introduce time jitter, they will have bigger or lesser influence cycle-to-cycle.

jlebar, 2020-04-25:
> For PID to run properly it has to be triggered from timer interrupt.

Turns out our PID library handles this.  To be perfect we might need to call it from the timer interrupt, but it's calling `millis()`, checking the time and so on.  So I don't think we need to add interrupts right now.

jlebar, 2020-04-25:
The source code for the PID library is actually pretty nice.  You can find it at `.pio/libdeps/uno/PID_ID2`.

Miceuz, 2020-04-26:
DOH! Right, I forgot this library takes care of that. Note though that currently we are on default 100ms, we never call SetSampleTime. My feeling is that this is too slow. The exact optimal PID execution time should be determined by phase margin wizard (who we still don't have?), but intuitively it should be slow enough to react to meaningful sensor readings after filtering, slow enough to provide meaningful command to actuator and fast enough to provide rise time we need. Some sources specify it at `< 0.1 * (Tblower - Tsensor)` where `Tblower` is the blower time constant and `Tsensor` is sensor filter time. Provided we can sample and filter sensor really fast compared to blower response `Tblower >> Tsensor` the good first guess is `0.1*Tblower'. Tblower can be measured by measuring step response of the blower, it's 63% of rise time when full speed command is applied. I believe our blower responds much faster than in 1 second.

Speaking of sensor sampling - to get meaningful data those also have to run on strictly stable clock so that quantization happens on equal intervals.

asmodai27, 2020-04-26:
Hi all
I have the feeling we are discussing several different topics here, which will all need adressing, but maybe not simultaneously:

1. the main "pid controller" (currently pid.cpp, tentatively renamed "respirationCtrl" in lee-matthews/issue103_controllerModule branch to avoid confusion), which sets the "target pressure", (and should manage the exhaust valve if I am not mistaken). This needs a lot of work (currently, nothing works in there), and is quite urgent as far as I understand our Alpha objectives.

2. the actual PID control loop, which activates the blower to reach "target pressure" and for which the code looks good to me (I do have access to aviation controls code to check it against, nothing sticks out), and its parameters need some tuning (sample time is one of the many parameters that need tuning), which will only happen when someone actually runs enough simulations to find proper parameters, but does not look as urgent to me.

3. the sensors, which will also need addressing, sooner rather than later to get decent readings, and for which at some point we will need to design and tune filters (including some prediction filters to have a reliable pressure assist mode), but maybe that is secondary for Alpha so long as we have reliable instantaneous readings?

jlebar, 2020-04-26:
> Speaking of sensor sampling - to get meaningful data those also have to run on strictly stable clock so that quantization happens on equal intervals.

Can you elaborate?  What's the failure mode if we don't read (say) the pressure sensor at regular intervals?

> currently we are on default 100ms, we never call SetSampleTime. My feeling is that this is too slow.

I share that feeling.

One thing I'd like me or someone to do is to move the external PID library into our code so we can modify it, see https://github.com/RespiraWorks/VentilatorSoftware/issues/174, which I just filed.  asmodai27@, it's nice to meet you -- this bug might be up your alley.

> I have the feeling we are discussing several different topics here, which will all need adressing, but maybe not simultaneously:

> the main "pid controller" (currently pid.cpp), which sets the "target pressure", (and should manage the exhaust valve if I am not mistaken). This needs a lot of work (currently, nothing works in there), and is quite urgent as far as I understand our Alpha objectives.

Correct, this is basically my outstanding https://github.com/RespiraWorks/VentilatorSoftware/pull/172.

> the sensors, which will also need addressing, sooner rather than later to get decent readings, and for which at some point we will need to design and tune filters (including some prediction filters [...]

Also correct.

closed 2020-04-26 by jlebar

Miceuz, 2020-04-27:
> > Speaking of sensor sampling - to get meaningful data those also have to run on strictly stable clock so that quantization happens on equal intervals.
> 
> Can you elaborate? What's the failure mode if we don't read (say) the pressure sensor at regular intervals?
> 

It's the same effect - when you are transitioning from continuous domain to discrete you have to keep the quantization timing constant in order to get all the frequency information right. Basically, if we don't read the value at regular intervals, we can't apply digital filtering to it. Depending on the frequency content I usually either have automatic ADC sampling set up (implemented differently in different MCUs) which interrupts with a new value on set frequency or just do 
```
if(isTimeToSample()) {
sample()
}
```
in the main loop if sampling is very slow compared to main loop speed.