RespiraWorks/Ventilator #18: Implement communication from Arduino to Pi

neelfirst, 2020-04-09:
Copied thread from inceptionev/FixMoreLungs

@rwilbur: We need to figure out if this is going to use UART or SPI

@lee-matthews: Indeed. Was planning on making possible to set those too. Maybe create some kind of "engineering/debug" mode on the interface module so that we can play with these values directly that's protected from a typical user.

@rwilbur Looking at the clock rate and from what people have said online it appears to be capable of 2Mbps. I agree the differential UART link would be nice to have because of the robustness to noise. I have used RS232 in quite noisy environments and have never had noise issues with it so in practice it might be perfectly acceptable. I think the fewer the parts the better.

@lee-matthews I presume the Arduino only has a UART output (0 to 3.3V) instead of +-12V? To add RS232, we may need to add an extra module anyway, a MAX232 perhaps? Maybe these come in premade modules for rpi / arduino.

@rwilbur I meant 3.3v when I said 232 but that is incorrect. I think the 3.3v UART would probably be fine. The UART on the PI is also 3.3v so they should play nicely together. If these 2 devices are super close together we can probably do a good job keeping noise down by just having a super connection.

neelfirst, 2020-04-11:
Note: for alpha this requires a voltage divider on the Arduino --> RPI line.

derekyerger, 2020-04-11:
I have experienced repeatable reliability issues in high-bandwidth serial comms between a RPi and 32u4-based Arduino platforms. I think this risk is worth mentioning, and should be accounted for in any preliminary testing for component and bus selection.

For low-bandwidth, occasional communication it works fine - I haven't managed to fault isolate but have 3.3v atmel 32u4's directly interfaced to a Pi Zero W for a project and do run into problems sending bursty data between the two.

This problem also spans other Arduino hardware and port choices. For example, SoftwareSerial should probably be avoided, and when used, the baud selection is highly coupled to clock speed.

toybuilder, 2020-04-11:
RPi and Arduino should have no problem communicating via UART at 3v3 CMOS logic-level swing. (As the signaling is internal to the product, we shouldn't need to use RS-232 level swing.)  Assuming the signal is routed properly, there should not be any noise related issues provided that the processors themselves are getting sufficiently clean power.

The "Arduino" needs to be clocked off a crystal/oscillator and not use the RC oscillator.  Provided that both the Pi and Arduino have accurate UART base clock, 2 MHz should be fine, as 2 MHz is an whole multiple of 48 MHz (RPi3), 30 MHz (RPi2) and 16 MHz (Arduino) base clocks.

SoftwareSerial is stupid and should never be used in real-time applications such as this.  Either interface to the hardware UART (i.e. interface through the RX/TX pins at the Atmel and have the USB interface in z-state) or go to a processor with additional UARTs (Mega).

Also, make sure that the Pi has the serial port in raw mode.  If there's any tty handling, it can slurp up data.  And, finally, make sure there is no data overflow.  At higher data arrival rates, there's not much time for the Arduino to work AND handle large amounts of fast incoming data.

Miceuz, 2020-04-12:
I have done RasPi - ATMega328 serial communication before. Having CRCs and frame markers helped a lot as there are still some packets lost from time to time.
It is tempting to have the protocol text-based, this way controller can be controlled directly by connecting to command line over serial - handy when testing. On another hand, maybe having something that has framing over UART makes more sense. Modbus?

martukas, 2020-04-13:
Blocked by #60

Miceuz, 2020-04-13:
BTW, what are bandwidth requirements for this channel? There is just a requirement for pressure display at 25Hz in req document.

lee-matthews, 2020-04-13:
It currently updates at 10Hz, but this can be changed easily.

martukas, 2020-04-23:
This is probably done?

closed 2020-04-23 by jlebar