RespiraWorks/Ventilator #2: Set up a custom disk image repository

twslankard, 2020-04-09:
I'm stuck on RespiraWorks/OpenSourceVentilatorGUI#13 because I need to modify the Raspbian image to make it more easy to automate VM-based testing the Qt-based GUI app.

In particular, `sshd` isn't enabled by default (which is peculiar, I know.) For now, all I need to do is run `sudo systemctl enable ssh.service` and reboot. I'm not a QEMU expert by any means, but it appears the disk image changes are persisted, so that's nice. I just need a place to save the modified image so that it's accessible to our Travis-CI instance and any developers.

hhff, 2020-04-11:
@twslankard should we just use dockerhub?

twslankard, 2020-04-11:
I think it depends on how we want to deploy our Docker images to the RPi. I imagine we'll have to flash them with a host OS+software that'll let us pull an image and manage the configuration and lifecycle of the containers. I was planning to build my own since I don't know what's out there. This isn't really my domain though - do you know of any off-the-shelf RPi disk images that'll do this?

twslankard, 2020-04-15:
@hhff any thoughts? I'm not sure whether this is needed at this point.

hhff, 2020-04-15:
@twslankard no - I don't I'm sorry. Was also planning to do it bespoke if I couldn't find anything.

But yep that was my plan too. Super lightweight RPi image that just pulls & runs docker containers. Should work without internet too, so we'll need to ship the latest docker image with that RPi image

twslankard, 2020-04-15:
Let me know if I can be of any help with this part. Curtis Kline hooked me up with a Linux instance (running in Paperspace's cloud) that we could use as a host, potentially.

hhff, 2020-04-15:
ok rad! thank you @twslankard