RespiraWorks/Ventilator #6: Display graphs for TV, Pressure, Flow

rwilbur, 2020-04-09:
# Requirement
The UI must be able to display the tidal volume, pressure, and calculated or measured flow rate in a graph format that updates in (near) real time

martukas, 2020-04-13:
Is this not already done, albeit with spoofed values?

martukas, 2020-04-13:
This possibly blocks #22

neelfirst, 2020-04-27:
https://github.com/RespiraWorks/VentilatorSoftware/wiki/Alpha-milestone has been updated with more specific guidance for what is needed by 5/1. I have thus assigned priority label for #4 #6 #8 #22.

jkff, 2020-04-29:
This is not done. This issue must build on https://github.com/RespiraWorks/VentilatorSoftware/issues/4 and involves:

- Having UI elements with these graphs (yes, we have that already)
- Binding data in these graphs to data obtained via the communication protocol (we don't have this)

jlebar, 2020-04-29:
Edited this bug to focus on just pressure and flow, these are the two that we're confident we need.

TODO @jlebar follow up whether we need tidal volume as a hard requirement for alpha.

jlebar, 2020-04-29:
From Ethan, TV is in the "very nice to have" camp.

neelfirst, 2020-04-30:
For TV - may be an unnecessary hack, but since we're only doing pressure command, can we just reset to zero every time?

PR #242 (2020-05-04): Adds GUI parameter controls and readouts
This is for the most part work by @lee-matthews - I took his branch and added finishing touches.

In the end:
* Adds controls for setting parameters, the parameters have reasonable default values, I confirmed that the ConnectedDevice gets them via debug logging
* Adds readouts of most recent sensor values. This competes with #235 (thanks @rwilbur! I should have communicated more clearly that I was doing this too) but I think the current UI looks nicer with Lee's parameter widgets cause it is symmetric.
* Sets graph Y ranges to reasonable values

Sorry f [... cut]
closed 2020-05-04 by jkff