RespiraWorks/Ventilator #8: Add TV, PEEP, PAW readout

rwilbur, 2020-04-09:
# Requirement
Must have 3 readouts displaying Tidal Volume, Positive End-Expiratory Pressure, and airway pressure in numerical format

martukas, 2020-04-13:
This possibly blocks #22

twslankard, 2020-04-25:
@rwilbur I don't see a PAW readout on [this mockup](https://respiraworks.slack.com/files/U010ZMKPT7W/F012S13H072/ventilator_ui.pdf). Do you know if this is the latest version?

neelfirst, 2020-04-27:
https://github.com/RespiraWorks/VentilatorSoftware/wiki/Alpha-milestone has been updated with more specific guidance for what is needed by 5/1. I have thus assigned priority label for #4 #6 #8 #22.

jkff, 2020-04-29:
Decision (@neelfirst): just display instantaneous values of whatever is on the graphs.

martukas, 2020-05-03:
Possibly blocked by #169

PR #242 (2020-05-04): Adds GUI parameter controls and readouts
This is for the most part work by @lee-matthews - I took his branch and added finishing touches.

In the end:
* Adds controls for setting parameters, the parameters have reasonable default values, I confirmed that the ConnectedDevice gets them via debug logging
* Adds readouts of most recent sensor values. This competes with #235 (thanks @rwilbur! I should have communicated more clearly that I was doing this too) but I think the current UI looks nicer with Lee's parameter widgets cause it is symmetric.
* Sets graph Y ranges to reasonable values

Sorry f [... cut]
closed 2020-05-04 by jkff