RespiraWorks/Ventilator #31: Ensure unit test reporting & test coverage reporting

martukas, 2020-04-10:


jkff, 2020-04-16:
As of #107, we have unit testing run via Travis. Don't have coverage yet.

hhff, 2020-04-17:
Currently working on setting up codecov

martukas, 2020-06-26:
@twslankard @a-vinod maybe we could circle back to this at some point in the next couple of weeks?

martukas, 2020-07-10:
This was partly addressed in #771 
Remaining sub-issues:
* completely untested files included in statistics
* test code itself ignored in statistics
* coverage of GUI

Additionally we could improve:
* link to html can be rendered as hyperlink in bash, so it is more convenient to view stats locally

And to cap it, publish results on either codecov.io or coveralls.io

tapemaster, 2020-07-29:
Test coverage is extracted to #840

closed 2020-07-29 by tapemaster