RespiraWorks/Ventilator #39: Reimplement PID control loop

martukas, 2020-04-10:


asmodai27, 2020-04-27:
For alpha the PID library has been imported locally to allow unit tests, and can more easily be tuned/fixed for Beta needs.
See https://github.com/orgs/RespiraWorks/projects/1#card-37049005

tapemaster, 2020-05-30:
@martukas @asmodai27 can you guys please write with more details what needs to be re-implemented here and why?

asmodai27, 2020-05-30:
I think this one can be closed actually, it was already taken care of when performing several other ones.
We still have to change the control laws from a simple PID to something different now we have up to 3 actuators in closed loop. That might be already covered with https://github.com/RespiraWorks/VentilatorSoftware/issues/378, https://github.com/RespiraWorks/VentilatorSoftware/issues/408 and https://github.com/RespiraWorks/VentilatorSoftware/issues/409

closed 2020-05-30 by asmodai27