RespiraWorks/Ventilator #50: Decide on algorithms to implement

martukas, 2020-04-10:


martukas, 2020-04-10:
Daniel Burnham appears to be working on this, but any additional help is very welcome.

neelfirst, 2020-05-08:
I don't think anyone is presently working on this. It seems we need serious algorithms for
- inspiratory effort detection
- respiratory rate measurement

As well as more basic ones like
- flow -> TV integration
- PIP detection
- I/E timing measurement
- PEEP detection

martukas, 2020-07-23:
Volume calc is #636 
Inspiratory detection related work is in #669 #388 #373 #374 #828
RR detection is #827
PIP #829
PEEP #830
I/E #831

martukas, 2020-07-23:
Additional algorithms might be needed for lung compliance detection.

martukas, 2020-07-29:
There needs to be a centralized place in repo that will be the source of truth for what those algs are. With precise definitions of those algorithms in narrative form.