RespiraWorks/Ventilator #51: Work on software architecture document

lee-matthews, 2020-04-10:


neelfirst, 2020-04-11:
@lee-matthews  I know you've put in a lot of work into this. Can you add a link here? What remains to be done? Can I get you some assistance? Thanks.

lee-matthews, 2020-04-13:
https://docs.google.com/document/d/1FPB31V72r_keu1_xjUfYCUXGLxBC5hoyCT1naPNNNTA/edit

Please feel to get someone to work on it :)

asmodai27, 2020-04-19:
Like I said on slack, I started a document on my end that describes the inner workings of the controller: https://docs.google.com/document/d/e/2PACX-1vQ5zpzNkm-p2jS8UI1QRwzOUAeX9tz00BAQlaYabDSOTT4NmYzgVcRgcdrfjCxeNCBpx-gVNb8adOCK/pub

If this is the kind of things you had in mind, feel free to add it to the architecture document.

tapemaster, 2020-07-29:
This has been worked on :P

closed 2020-07-29 by tapemaster