RespiraWorks/Ventilator #55: Check used PID library - check reliability. Has it been unit tested?

lee-matthews, 2020-04-10:


neelfirst, 2020-04-11:
Possible duplicate of #39

asmodai27, 2020-05-01:
Currently playing with PID lib in test mode, I see one major issue with the way it keeps track of time and assumes constant sample time to compute integrals and derivatives:

When I introduce a jitter in the task, the PID computes the integral and derivatives assuming it was called with exact sample time (dInput = (input - lastInput) which is then multiplied by kd = Kd/SampleTime in output), but records start of execution time (with millis()) as lastTime and will only refresh output when millis() > lastTime + sampleTime, and compute the derivatives and integral assuming constant SampleTime, therefore having "lost" a fragment of time in those integral and derivatives.

I see several steps in fixing this, from easiest to more complicated (and of course each step brings more robustness in the PID):
1. use lastTime += SampleTime instead of lastTime=now (will also require proper init on first call), quick fix that makes the PID not lose fractions of time but integral and derivatives are still not perfect due to jitter,
2. compute derivative and integral using actual time between calls instead of constant sample time, this poses the problem that call time is not actually the time of the reading, introducing a small error in those as well,
3. perform derivation and integration faster than SampleTime (which is actually desired refresh time, not actual sample time), this will require more complex algorithm for derivative which may or may not yield better estimations (in the case of the derivative, the integral should be better if performed faster). This one is definitely not mandatory and may be a little ressource-heavy.
4. have the sensors timestamp their readings and use those timestamps to compute integral and derivatives.

I'll start working on 1 and 2, and possibly the other as well depending on the time I have on my hands.

closed 2020-05-04 by asmodai27