RespiraWorks/Ventilator #56: Develop EEPROM code to store and retrieve data

lee-matthews, 2020-04-10:


martukas, 2020-04-10:
Is this duplicate with #15 ?

martukas, 2020-04-14:
@neelfirst: is this critical for Alpha?

jlebar, 2020-04-25:
Most/all params should be stored in flash and restored on restart.

jlebar, 2020-04-25:
Open question here is whether the storage needs to be retained across power loss, or just across controller restarts.  Sort of seems like, if we lose power, we're screwed anyway.  OTOH if we lose power *very briefly* we don't want to interrupt care.

verityRF, 2020-04-25:
FWIW I don't think this is a real system requirement, but a perceived user requirement. 
Losing power as in, losing mains, is to be expected which is why a battery backup is planned. The switchover can happen in such a way that there is no loss of power to the board. 

Now if power to the board _is_ lost actually, what happens next is a bit muddled. Do we really think that by the time they restart it, the old parameters before power outage would be the same? The nurse would have to monitor the patient and dial in the new settings most likely due to patient stress/changes in time. 

For dev builds this seems not terribly useful either since, well, it's a dev environment. 

If we go this route, we have a whole bunch of other failure analyses to run and think about. Now we need checksums for the flash contents. What if it was corrupt? What if we hit the endurance of the max writes in the field? What if the last parameter save didn't finish? Do you replace the whole unit since the IC is soldered down? How would it notify the nurse that the flash is degraded, etc, etc. If we don't it's a lot less on our plate.

My 2 cents anyway.

jlebar, 2020-04-25:
Busy at the moment, but if you have 5m, check out the Medtronic PB560 manual?  I think it explicitly talks about this, but I can't remember their decision.

I think your points are all valid -- although this is all speaking against writing to non-volatile memory that survives power loss.  We still need to survive a restart (right?).

verityRF, 2020-04-25:
Interesting.
According to their user manual, alarms logs, and errors are logged to nonvolatile memory.

If the unit was in ventilate mode, but the power switch was hit, it'll start up in ventilate mode. Last used settings recalled, unless there was a fault in the memory, then it'll alarm the nurse and ask to reset the parameters. 

Interestingly, they don't mention if power was completely lost, whether the unit will start up in ventilate mode or not. Just if the physical power switch was used.

verityRF, 2020-04-25:
And I'm not sure what survive a restart means to you...The program memory is non-volatile so it would just be any old power on reset and the ventilator would start in some well defined idle state. Whether the dynamic user applied parameters are saved to some flash is the question...but the system doesn't need them to enter a safe idle state (since defaults would be programmed in to begin with).

jlebar, 2020-04-25:
> And I'm not sure what survive a restart means to you

I mean, if the controller resets due to (say) a watchdog timeout, should it enter an idle state, or should it start back ventilating with the last settings?

Maybe our answer is, idle state, it'll get state from the GUI very soon.  Unless the GUI is *also* dead at the moment...

verityRF, 2020-04-25:
I'll have to dig more into the Medtronic stuff to see how they handle a non power-switch related failure.

sglow, 2020-04-25:
I don't have a strong opinion on the high level pros and cons of saving the settings during run time.  I do have some thoughts about low level implementation if we do decide to save parameters to flash.
One simple method that I've used successfully for this is to reserve two flash pages for parameter storage.  On the STM32 flash pages are 2k bytes, so that would be 4k of flash reserved for parameter storage.
The parameters themselves are stored in a structure with a fixed size less then the flash page size.  Ideally some integer multiple of these structures can fit within one flash page.  Say 128 bytes for argument sake.
In addition to the parameter values the structure has a 32-bit CRC, a counter byte and a 'mark' byte used to ID a good flash block.
To save parameters, write a new structure to the next free flash location, erasing a page if necessary.  The counter value is the counter value of the current valid parameter block +1, the mark is some fixed value (not 0xFF or 0x00), say 0x55.  The CRC is calculated based on the contents of the block.
After writing to flash, read back and verify the write was successful.  If so, write 0 to the mark on the old block of flash.  You can write zeros to flash even if some bits have already been cleared.
On startup, scan through the two pages looking for valid blocks.  Valid blocks are ones with the correct mark value and valid CRC.  There should be at least one.
If there are two good blocks (could happen if you lost power during a save), use the one with the higher counter value obviously taking the wrap from 255 to 0 into account.
If there are no valid blocks then create one with reasonable default values.  That should only happen on a new system.

sglow, 2020-04-25:
One possible issue with writing to flash on the STM32 is that we will almost certainly be running our code from flash at the same time.  You can't read from flash at the same time you're writing or erasing.
The STM32 actually handles this very simply; it stalls the processor if you try to read from flash while a write or erase is taking place.  As long as you don't do something stupid like erase the block that your code is executing from this all works just fine.  The only possible issue is that the processor stops running briefly.
Programming 64-bits of flash (the smallest amount you can program at one time) takes about 90 usec.  Erasing a 2k page of flash takes about 25 ms.
This means that if we are going to be storing parameters to flash while the system is running we will have to anticipate the processor stalling for up to 25ms during the page erase.  That's generally not a good thing if we're running servo loops.  It will also cause loss of serial data, etc.
There are ways around this.  We could run all the critical code out of RAM so it's not effected, but that has downsides.  It uses up a lot of RAM and you have to be very careful about what code could possibly be run within the critical loops.  Not just the loop code itself, but any library function that the loop code could call, etc.
The other option is to choose an STM32 chip that has two flash banks.  The one we're currently targeting doesn't have that, but we could always switch chips if we determine that storing parameters to flash is a hard requirement.
I guess there are other options as well, we could use an external serial flash.  Also, I think some STM32 chips have the option for a small battery backed RAM.  I'm not too familiar with that last option.
I guess the point is, storing parameters to internal flash may be more trouble then it's worth.

sglow, 2020-04-25:
One more note, sorry.  I just noticed issue #97.  We will definitely need to be able to store things like calibration data to flash, so some method like the one I outlined above is definitely needed.  As long as we don't save anything to flash while the ventilator is running the stall issue shouldn't matter.  Storing calibration data is something that would only be done during system commissioning, never when the ventilator is actually in use.

hobodrifterdavid, 2020-04-25:
I implemented a 'stack' in the STM32 4k backup ram for storing 'records', using STM32 HAL. Code looks like this, if you are curious:
`// Some code to store "records" (binary blobs) in the 4k backup SRAM of the STM32.
// Records are stored in a stack. You can push a record onto the stack, or pop one off.
// So it's FIFO (first in, first out). There's also CRC to make sure we are popping
// valid records.
// Made by David.

#include "global.h"
#include "stm32f4xx.h"
#include "backup.h"

uint8_t* peekPointer;
uint16_t crc16(const uint8_t* data_p, uint8_t length);

void backup_init(void) {
	// Enable backup SRAM
	/*
	RCC_APB1PeriphClockCmd(RCC_APB1Periph_PWR, ENABLE);
	PWR_BackupAccessCmd(ENABLE);
	RCC_AHB1PeriphClockCmd(RCC_AHB1Periph_BKPSRAM, ENABLE);
	PWR_BackupRegulatorCmd(ENABLE);
	*/
	
	    /*PWREN : Enable backup domain access  */
    __HAL_RCC_PWR_CLK_ENABLE();
	   /*DBP : Enable access to Backup domain */
    HAL_PWR_EnableBkUpAccess();

		__BKPSRAM_CLK_ENABLE();
    /*BRE : Enable backup regulator
      BRR : Wait for backup regulator to stabilize */
    HAL_PWREx_EnableBkUpReg();
	

   /*DBP : Disable access to Backup domain */
    //HAL_PWR_DisableBkUpAccess();
}

// Reset the stack 'freePointer' so that the stack is empty
void backup_clear(void) {
		*(void**)BKPSRAM_BASE = ((uint8_t*)BKPSRAM_BASE) + 4;
		peekPointer = ((uint8_t*)BKPSRAM_BASE) + 4;
		printf("BACKUP: Stack reset.\r\n");
}

// Store a record of up to 256 bytes
bool backup_push(void* dataPointer, uint8_t sizeBytes) {
	
	/* Enable clock to BKPSRAM */
  //__HAL_RCC_BKPSRAM_CLK_ENABLE();
	
	uint8_t* freePointer = *(void**)BKPSRAM_BASE; // get current stack pointer
	
	if(freePointer < (((uint8_t*)BKPSRAM_BASE) + 4) || freePointer > (((uint8_t*)BKPSRAM_BASE) + 4095)) {
		// freePointer is outside of the valid range.
		printf("BACKUP: freePointer is outside of the valid range (%p).\r\n", (void*)freePointer);
		backup_clear();
		freePointer = ((uint8_t*)BKPSRAM_BASE) + 4;
	}
	
	if( (freePointer + sizeBytes + 3) > ((uint8_t*)BKPSRAM_BASE) + 4095) {
		// error, insufficient space
		printf("BACKUP: insufficient space for record.\r\n");
		
		//__HAL_RCC_BKPSRAM_CLK_DISABLE();
		return false;
	}
		
	// We can write
	
[... 179 lines cut ...]
	
	if(calcCRC == savedCRC) {
		return true;
	}
	else {
		printf("BACKUP: WARNING: CRC failed for value in backup slot %d.\r\n", slotNo);
		return false;
	}
}

void backup_put(uint8_t slotNo, uint32_t value) {
	
	if(slotNo > 9) {
		printf("BACKUP: ERROR: There are only 10 backup slots, slot no. %d requested.\r\n", slotNo);
	}
	
	HAL_RTCEx_BKUPWrite(&hrtc, RTC_BKP_DR0 + slotNo, value);
	
	uint16_t calcCRC = crc16((uint8_t*)&value, 4);
	
	HAL_RTCEx_BKUPWrite(&hrtc, RTC_BKP_DR0 + slotNo + 10, calcCRC);

}
	
`

jlebar, 2020-04-25:
> One possible issue with writing to flash on the STM32 is that we will almost certainly be running our code from flash at the same time. You can't read from flash at the same time you're writing or erasing.
The STM32 actually handles this very simply; it stalls the processor if you try to read from flash while a write or erase is taking place. As long as you don't do something stupid like erase the block that your code is executing from this all works just fine. The only possible issue is that the processor stops running briefly.
>
> Programming 64-bits of flash (the smallest amount you can program at one time) takes about 90 usec. Erasing a 2k page of flash takes about 25 ms.

Ugh, this is a problem.  90us is probably fine, but I think 25ms is a long time for us.  Or at least, we are driving a PID loop with a cycle time that (I expect?) should be much faster than that.  As you say, we can try to ensure that all of that code is in RAM, but that seems extremely easy to mess up, inasmuch as if any part of that critical path lives on flash, it'll work fine in basically all testing (right?).

(As a brief postmortem, I'd thought about this and realized that the 25ms erase time would be slow. 
 But I hadn't realized we wouldn't be able to read from flash while erasing a page, so I thought we could hide the latency.)

> As long as we don't save anything to flash while the ventilator is running the stall issue shouldn't matter. Storing calibration data is something that would only be done during system commissioning, never when the ventilator is actually in use.

Right.  Now the question of whether we need to store non-calibration data to non-volatile memory becomes more crucial.

It sounds like we should ask the hardware team whether we can get a chip with two flash banks, or if that decision is fixed at this point.  @sglow do you have a chip to recommend?

Miceuz, 2020-04-25:
hm. can we hide the latency by doing DMA write to an external (Q)SPI flash?

toybuilder, 2020-04-25:
In addition to external EEPROM, look into either battery/capacitor backed SRAM (including RTCs with scratchpad registers if you only need to store a little bit of data) or look into FRAM (F-RAM, FeRAM).  http://www.ti.com/lit/ml/slat151/slat151.pdf?ts=1587831357701 as they do not have the same erase delays.

sglow, 2020-04-25:
I think the first and most critical question is whether we actually need to store parameters during normal operation.  If so, exactly what parameters and why?  If it turns out that we don't, then this is all a non-issue.
Storing calibration data is a different matter.  That seems like something we'll definitely need, but I don't think the CPU stalling issue is really a factor here.  The unit would never be calibrated while in use.  Maybe we have a safe state that the unit can be put in basically meaning it's not being used and the servo loops are idle and only allow parameters to be updated when in that safe state.
If it is determined that we need to store parameters during normal operation then I think we should either select a chip with dual flash banks, or add a small SPI flash.  I don't think either one is a major architectural change at this point.
Lets answer the first question of whether or not saving parameters during normal run time is necessary first and only try to solve the problem of how to do it if it's determined that it's necessary.

sglow, 2020-04-25:
I'll post the question to the system design channel on Slack

inceptionev, 2020-04-25:
Doesn’t seem to me that adding a small external flash IC would be a high barrier.  These things cost a couple dollars and we’re spinning a PCB anyways.  For the alpha PCB, there is a microSD card accessible from the STM32.

inceptionev, 2020-04-25:
I’m also not opposed to switching to a different STM32 with dual banks.  We might also want that to guard against firmware corruption by having two copies of the firmware and a bootloader that can switch between them if one fails.

esc-works, 2020-04-25:
Edwards answer, for what its worth:
> We did design in external non-volatile memory. However, we also don’t plan (at least so far) to write NV memory during operation. In addition, we plan to have the Raspberry Pi keep track of most non-volatile data, which removes the need for the MCU to perform this expensive write operation. Also, in general, we don’t plan to keep ventilator state between power-cycles. It is safer to always initialize the vent to the same consistent state on power-up. We will have non-volatile calibration data, version information, etc., but this is expected to be static data programmed at the factory or on system upgrade. Hope this helps.
> -Edward

Miceuz, 2020-04-25:
I think it will be dumber, safer and easier for us to fail with alarm on MCU reset. It will be much more complicated to dance around various issues with "wake up in the middle of crisis" situation both in code and in risk analysis. Reset must NOT happen. We deal with it but not try to recover fully.

We must incorporate this fact in our risk analysis though.
|||
|------|----|
|Risk identification |Controller unit resets during patient treatment|
|Risk effects | Death, severe health damage |
| Severity | 10 |
| Risk root analysis | Failure in the controller unit firmware or in electrical system of the controller caused the microcontroller reset. |
| Likelyhood | 6 |
| Risk priority number | 10 * 6 = 60 |  
| Risk elimination possibilities |Attract attention of medical personel by issuing HIGH level alarm. Detect the unexpected reset in firmware and fail safely, alter the common startup procedures.
|Risk limitation actions done | [list of closed github issues dealing with this]
|After conducted risk elimination activities | 
|Severity |  4
|Likelyhood | 4
|Risk priority number| 4*4 = 16

verityRF, 2020-04-25:
Sounds like we don't have a need for saving the state in flash...for calibration we can just use the flash as-is since it's a factory thing.

jlebar, 2020-04-27:
As an update, @verityRF and I talked today and concluded that we *may*
in fact need to save calibration data in non-volatile memory in the
field.  The issue is that we may need to recalibrate the pressure
sensors due to changes in temperature, humidity, someone dropping the
device, whatever.

That said, we're pretty sure we can't calibrate while a patient is
hooked up, so it's nbd if calibration causes the device to hang for a
second or whatever.

neelfirst, 2020-04-27:
We have an SPI interface for a microSD card for this rev so at least SW dev can proceed unhindered. We can replace with such a chip for prod.

Relates to: https://github.com/RespiraWorks/pcbreathe/issues/54

jlebar, 2020-04-29:
There's a lot of discussion here (thanks, everyone), and I think we should forward-dupe this to #170, which covers the one use of non-volatile storage that I think we all agree we need.

It's conceivable we'll need non-volatile storage for something other than storing calibration data, too.  We might want to store the ventilation configuration itself, although @verityRF and @Miceuz both argue that maybe we shouldn't.  We might also want to have two copies of the microcontroller code so that if an update fails we can fall back to a known-good state.

All good; we can hash that out in separate bugs or (for some of these) in discussions about requirements.

jlebar, 2020-04-29:
Closing as a dupe of #170

closed 2020-04-29 by jlebar