RespiraWorks/Ventilator #57: Add sensor code

lee-matthews, 2020-04-10:
- Add ability to read from flow and pressure sensors
- Get calibration data
- Apply calibration data
- Provide warnings if sensor failure detected

martukas, 2020-04-10:
Could it be that this is duplicate and/or already broken down into smaller tasks in #35 #36 #38 #40 #41 #44 ?

neelfirst, 2020-04-11:
I concur with @martukas , issues should be kept focused and small, so related PRs can be similarly small and reviewed quickly.

neelfirst, 2020-04-11:
This also seems like controller specific code, rather than common code. I'm not sure the raspi has any business interfacing with sensors.

martukas, 2020-04-14:
Blocked by #41 #40 #44

lee-matthews, 2020-04-14:
Indeed, this code is destined to manage the sensors on the Arduino board.

PR #74 (2020-04-15): Pressure Sensor Code Integrated + Arduino Unit Test Start
Implemented controller module for getting pressure readings from the 3 pressure sensors used in the ventilator design as per the HW ICD. Wrote basic unit test.

Wrote first steps for Arduino unit testing. Created fake Arduino wrapper so unit tests can be run independently of the Arduino. This can be reused and added to for further unit testing of other modules.

Module succesfully passed unit test. Can use module as part of a simulation for the master state machine without needing to run it on an Arduino.

 Closes #57
commit 512eb59b (2020-04-15): Merge pull request #74 from RespiraWorks/issue_57_sensorcode

Pressure Sensor Code Integrated + Arduino Unit Test Start. Closes #57 (good enough for alpha)
closed 2020-04-15 by verityRF