RespiraWorks/Ventilator #58: Parameter module work

lee-matthews, 2020-04-10:
- Encapsulation setter functions to ensure config data is within hard limits
- Add framework for storing and retrieving calibration data

neelfirst, 2020-04-11:
Is there more to this ticket than #15 and #17 ? If so let's try to distinguish it further. Thanks.

lee-matthews, 2020-04-13:
I was thinking of the following. Use the parameters module to store and retrieve that data that is accessed frequently (reading from eeprom is SLOW), then the parameters module can occasionally store data in eeprom. When setting a mode (before starting the ventilator for example) we could save those parameters in eeprom so that we can automatically recover them in case of an arduino  reboot. In my opinion we should use the eeprom is as little as possible because they degrade with write cycles and they're slow.

martukas, 2020-04-13:
Can we split this into two tickets, roughly like the items identified in description? And it seems like the first might be blocked by #17? Or is it synonymous with #17?

lee-matthews, 2020-04-16:
I'm closing this because I believe it is covered by other issues.

closed 2020-04-16 by lee-matthews