RespiraWorks/Ventilator #60: Factor out communications protocol code into common directory

martukas, 2020-04-10:


neelfirst, 2020-04-11:
Related issues: #4 #18

neelfirst, 2020-04-11:
Blocked by #59

martukas, 2020-04-13:
This is no longer blocked, and it looks like it might even be done? Or is there still work to do?

jlebar, 2020-04-13:
> looks like it might even be done?

At 31a1fa9df703b7619874c15442e9f3c632b844d2, the comms code is still in controller/lib/comms.cpp, so I wouldn't say this is done, at least as I conceived this task.

lee-matthews, 2020-04-13:
I've moved the checksum code into the common/libs/checksum directory. I've placed the packet definition enums into the common/include directory. Not sure what other comms code is relevant to move into the common directory. I can have a look if you like?

jlebar, 2020-04-13:
> Not sure what other comms code is relevant to move into the common directory.

I guess I was imagining a system in which all of the message structures (e.g. "4 bytes of X, 8 bytes of Y, followed by 2 bytes of padding") were defined in one place in the code.  So the serialization and deserialization logic for a particular message live right next to each other (or, ideally, are generated from the same file).  Then the GUI side just plugs into this.

Miceuz, 2020-04-13:
Shouldn't the comm code be split to a separate repo eventually? This way we could control its version and ease the decoupling of GUI and controller.

jlebar, 2020-04-13:
We decided to go with the "monorepo" approach instead.  We don't want the controller and GUI to be decoupled; they will always be built from the same git source.

martukas, 2020-04-14:
Yes, we will consider this done when serialization & deserialization code will be in in one place. Whether this is done by library or whatever, we need this by the end of the week. If we later need to write something that is more optimized, we can come back to this. Time is more expensive than memory.

Miceuz, 2020-04-14:
I'll add nanopb

lee-matthews, 2020-04-14:
Please wait on that Albertas. We've already had that discussion and we
decided against this. The discussion is on slack if you want to go and have
a look. Adding this would have ramifications for both the controller and
the GUI.

rwilbur, 2020-04-14:
I think this is as good of a place as any to discuss this since it will be well documented. What to we see the advantages of nanopb being? I know it is a serialization library but we will still need to send that information across in a protocol of some sort to get that across the communication bus right?

jlebar, 2020-04-14:
> I know [nanopb] is a serialization library but we will still need to send that information across in a protocol of some sort to get that across the communication bus right or am I misunderstanding something?

You're understanding 100% correct.  nanopb is just serialization, not transport.

> What to we see the advantages of nanopb being?

Right now there are two pieces to our communication protocol.

 1. How we send bits between the devices.  This is basically "the serial bus", but there's still work to be done here, e.g. Albertas has mentioned message framing (so that a if we corrupt a message's `length` we can recover) and also avoiding collisions on the serial bus.

  2. How to interpret the bits we receive.  This is basically Lee's document, https://docs.google.com/document/d/1caNhNaAoA6F9b-_AuyL1Mq6jvu-hL6lDtQXxJmiI1K8/edit

Right now this interpretation is manually coded.  When we send a message, we construct a `char[]` put bits into it, and then send that on the serial bus.

With nanopb, we would first write a `.proto` file which defines each of the messages.  It's basically just Lee's document from above translated into code.  nanopb can then translate this `.proto` file into C code; basically, a bunch of structs.  When we want to send a message, we construct one of these structs, ask nanopb to serialize it, and then send those bits on the wire.

The main reason to do this is because right now if you want to send or receive a message on the GUI, you're going to need to implement the other side of `controller/src/comms.cpp`.  It's possible to do it, for sure.  But it's a lot of code, and it's easy to screw it up, especially when someone wants to make a change later on.

Once we have this set up, the task of "let the GUI send/receive messages" becomes trivial, and it will never be out of sync with the controller.

If we decide nanopb isn't suitable for our use-case, I think we should probably write our own serialization library.  For our use-case it's not that hard.  But before doing that, I wanted to try the off-the-shelf tool.

lee-matthews, 2020-04-14:
Thanks for the info Justin.

This isn't a question of blame, but rather one of consent.

Technically, I don't believe this solution will add anything, for reasons I've already stated. But more important than this, and for me this is worse, is the lack of consultation. Riley and myself have already invested a lot of time in this, it would have been better to speak to us before making this decision. I appreciate that you say that this is not a decision, just a test, ok I understand this, but I sincerely believe that energies could be invested more wisely elsewhere. 

Again, I just want to state why I don't think this is a good idea.

1) Our communications protocol is simpl*, the average packet length is about 6-10 bytes long, the maximum length at the moment is 21 (I think) bytes.
2) We don't send complex data types - a few floats, timestamps, integers.
3) Adding nanopb will meaning adding tens of kilobytes of code which we have little control over and that we would need to rely on others to test.
4) Just the nanopb code itself uses about 300 bytes of SRAM, then on top of this one would need to add the structures to the SRAM requirements.
5) Even with nanopb, we would still need to handle error checking, re-transmissions AND encapsulate the serialized packets into another packet that adds length, checksum and type information.
6) Our communications protocol, whilst it hasn't undergone stress testing, works. 

We could use a bigger processor, I'm in agreement with this, but only if it is strictly necessary - if the main ventilation algorithm requirements are too demanding for the microprocessor for example. I think adding nanopb is overkill for what we are trying to achieve, more than that it adds another set of unknowns.

Personally I think we need to work on our decision making process.

I understand people are curious and that they want to build the best system that they can, it's commendable, I just worry that by doing this we are diluting our energy to places that don't require it.

jlebar, 2020-04-14:
I want to acknowledge that this was painful for you, @lee-matthews.  I also want to acknowledge that you have written the vast majority of this code.  You're the number-one contributor, and so your opinion matters a lot.  I'm still very new here.

I'm sorry, and I will strive to do better.

> I just want to state why I don't think this is a good idea.

I agree with pretty much everything you wrote.  No argument.

My main reason to want to try it is that I think it may get us working GUI communication more quickly.  That's all.

rwilbur, 2020-04-14:
I don't want anyone to feel bad about raising concerns or trying things out. No doubt we have some inefficiencies in communication and development. I know our communications protocol is very small and simple so I am still trying to wrap my head around what exactly it can help with. Please don't feel like you can't explore this I just am curious how it can help.

lee-matthews, 2020-04-15:
I agree with Riley, nobody needs to feel bad. It was a communiations mixup,
these things happen. We're a new team, mixing part time and full time,
spread over the globe, it's not surprising that things like this happen.

Miceuz, 2020-04-15:
So how do we move forward? 

I propose this track:
 1. Close this and #85 as won't fix 
 2. Create tickets for solving framing problems
 3. Create tickets for solving bus arbitration problems
 4. Create tickets for GUI to decode packets coming from controller

jlebar, 2020-04-15:
Talked to Lee offline, he said he's OK with us continuing with this
experiment, especially because

 - the GUI side doesn't have any serialization code written yet, and
 - per the discussion in #microcontroller-selection, it seems likely
we're going to get a more-powerful chip.

Miceuz, 2020-04-23:
Can we also close this now?

closed 2020-04-23 by jlebar