RespiraWorks/Ventilator #66: Optimize checksum code and write tests

jlebar, 2020-04-11:
I think the checksuming code can be improved somewhat: https://gcc.godbolt.org/z/cHmBLE.

More this likely-unimportant optimization, I want to use this as an excuse to write tests and set up testing infrastructure.

jlebar, 2020-04-11:
There's also some refactoring to do, as part of the logic of `checksum_check` is duplicated in `serialIO.cpp`.

jlebar, 2020-04-11:
See also https://en.wikipedia.org/wiki/Fletcher%27s_checksum#Optimizations

martukas, 2020-04-13:
Is this urgent enough to be in alpha?

jlebar, 2020-04-13:
Unit tests and testing infrastructure is highly urgent, yes!

martukas, 2020-04-13:
Yes, unit testing is urgent, I agree :) I was more questioning the urgency of optimization. But all is good.

closed 2020-04-16 by jlebar