RespiraWorks/Ventilator #75: blower control module

verityRF, 2020-04-13:


lee-matthews, 2020-04-13:
I'm thinking about changing the blower module to an actuator module, because there are other outputs too. A solenoid, and perhaps temperature control for the humidifier - what do you think?

lee-matthews, 2020-04-13:
Never mind previous comment. I think I'll create a solenoid module, we could then maybe create an abstraction layer above that if necessary.

martukas, 2020-04-14:
Make a ticket for the abstraction and put it in beta.

jlebar, 2020-04-23:
pid.cpp controls the blower.  We might want to refactor, but does the fact that we have a module which controls the blower mean this is done?

verityRF, 2020-04-23:
pid.cpp also had internal manual ways of creating sensor data, which is now put into a proper sensors.cpp. Blower needs to be put into its own module (which is still empty according to master right now). This is because other modules (aka sensors.cpp) have a dependency to be able to control/disable the blower to function 100% properly. 

I think lee was maybe working on this? Anyway it doesn't seem that hard since all it seems to involve is a single analogWrite.

jlebar, 2020-04-26:
>  Blower needs to be put into its own module (which is still empty according to master right now). This is because other modules (aka sensors.cpp) have a dependency to be able to control/disable the blower to function 100% properly.

Got it.  I think it would be better to have a bug describing the functionality that's currently missing from those other modules -- if that's OK with you?  Then we can triage whether those  other bugs block alpha.

Going to close this; do you mind filing those other bugs (since I'm not sure what functionality is missing)?  If you disagree, feel free to reopen, no hard feelings, just trying to clean things up.

closed 2020-04-26 by jlebar