RespiraWorks/Ventilator #79: Create solenoid control module

lee-matthews, 2020-04-13:


martukas, 2020-04-14:
@neelfirst @lee-matthews @jlebar  is this blocked by anything? Does any other feature need this?

lee-matthews, 2020-04-14:
No. The control loop will need this to control the solenoid. I've coded the
module already, it's on GitHub.

martukas, 2020-04-14:
Ah, ok. I see the branch.
It would be very helpful if you could tag your commits with the issue #, as suggested [here](https://github.com/RespiraWorks/VentilatorSoftware/wiki/Branch-naming-&-commit-messages), which will automatically update the ticket.

lee-matthews, 2020-04-14:
That's useful. Thanks. I'll do that from now on.

neelfirst, 2020-04-14:
Just so we're clear, the solenoid doesn't need any complex control, it is a simple digital I/O that feeds a FET driver to turn on/off.

lee-matthews, 2020-04-14:
Exactly, the solenoid would be controlled from the ventilator control algo
that hasn't been done yet.

jlebar, 2020-04-23:
solenoid.cpp exists.

closed 2020-04-23 by jlebar