RespiraWorks/Ventilator #101: Gather, timestamp, and transmit sensor data to interface computer

lee-matthews, 2020-04-16:
Read flow, pressure and zone information from the respiration cycle state machine, add timestamp and transmit to GUI interface machine.

jlebar, 2020-04-23:
This is a matter of fixing in the following TODO: https://github.com/RespiraWorks/VentilatorSoftware/blob/fa15b89/controller/src/pid.cpp#L126

The issue is I'm not sure what to put in there.  I'm planning to go to Oakland this weekend to touch the physical device.

jlebar, 2020-04-25:
This issue is not yet closed per: https://github.com/RespiraWorks/VentilatorSoftware/pull/160#issuecomment-618938322

jlebar, 2020-04-26:
Renaming based on what's left to do here.  Volume isn't needed for alpha.

jkff, 2020-04-29:
This is a dupe of other bugs related to calculating TV.

closed 2020-04-29 by jkff