RespiraWorks/Ventilator #112: FullScaleReading test is failing at master

jkff, 2020-04-16:
To reproduce:
Remove `DISABLED_` in `SensorTests.cpp` next to the test.
Run `pio test -e native`

Output includes errors:
```

Processing sensor_tests in native environment
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Building...
Testing...
Running main() from gmock_main.cc
[==========] Running 1 test from 1 test suite.
[----------] Global test environment set-up.
[----------] 1 test from SensorTests
[ RUN      ] SensorTests.FullScaleReading
controller/test/sensor_tests/SensorTests.cpp:147: Failure
The difference between pressureInhalation and differentialFlowPressures[index] is 2, which exceeds COMPARISON_TOLERANCE, where
pressureInhalation evaluates to 0,
differentialFlowPressures[index] evaluates to -2, and
COMPARISON_TOLERANCE evaluates to 0.004999999888241291.
Differential Sensor Calculated Value at index 4
controller/test/sensor_tests/SensorTests.cpp:147: Failure
The difference between pressureInhalation and differentialFlowPressures[index] is 1.5, which exceeds COMPARISON_TOLERANCE, where
pressureInhalation evaluates to 0,
differentialFlowPressures[index] evaluates to -1.5, and
COMPARISON_TOLERANCE evaluates to 0.004999999888241291.
Differential Sensor Calculated Value at index 6
controller/test/sensor_tests/SensorTests.cpp:147: Failure
The difference between pressureInhalation and differentialFlowPressures[index] is 1, which exceeds COMPARISON_TOLERANCE, where
pressureInhalation evaluates to 0,
differentialFlowPressures[index] evaluates to -1, and
COMPARISON_TOLERANCE evaluates to 0.004999999888241291.
Differential Sensor Calculated Value at index 8
controller/test/sensor_tests/SensorTests.cpp:147: Failure
The difference between pressureInhalation and differentialFlowPressures[index] is 0.5, which exceeds COMPARISON_TOLERANCE, where
pressureInhalation evaluates to 0,
differentialFlowPressures[index] evaluates to -0.5, and
COMPARISON_TOLERANCE evaluates to 0.004999999888241291.
Differential Sensor Calculated Value at index 10
controller/test/sensor_tests/SensorTests.cpp:147: Failure
The difference between pressureInhalation and differentialFlowPressures[index] is 0.5, which exceeds COMPARISON_TOLERANCE, where
pressureInhalation evaluates to 0,
differentialFlowPressures[index] evaluates to 0.5, and
COMPARISON_TOLERANCE evaluates to 0.004999999888241291.
Differential Sensor Calculated Value at index 14
controller/test/sensor_tests/SensorTests.cpp:147: Failure
The difference between pressureInhalation and differentialFlowPressures[index] is 1, which exceeds COMPARISON_TOLERANCE, where
pressureInhalation evaluates to 0,
differentialFlowPressures[index] evaluates to 1, and
COMPARISON_TOLERANCE evaluates to 0.004999999888241291.
Differential Sensor Calculated Value at index 16
controller/test/sensor_tests/SensorTests.cpp:147: Failure
The difference between pressureInhalation and differentialFlowPressures[index] is 1.5, which exceeds COMPARISON_TOLERANCE, where
pressureInhalation evaluates to 0,
differentialFlowPressures[index] evaluates to 1.5, and
COMPARISON_TOLERANCE evaluates to 0.004999999888241291.
Differential Sensor Calculated Value at index 18
controller/test/sensor_tests/SensorTests.cpp:147: Failure
The difference between pressureInhalation and differentialFlowPressures[index] is 2, which exceeds COMPARISON_TOLERANCE, where
pressureInhalation evaluates to 0,
differentialFlowPressures[index] evaluates to 2, and
COMPARISON_TOLERANCE evaluates to 0.004999999888241291.
Differential Sensor Calculated Value at index 20
controller/test/sensor_tests/SensorTests.cpp:154: Failure
The difference between pressurePatient and patientPressures[index] is 0.5, which exceeds COMPARISON_TOLERANCE, where
pressurePatient evaluates to 0,
patientPressures[index] evaluates to 0.5, and
COMPARISON_TOLERANCE evaluates to 0.004999999888241291.
Patient Sensor at index4
controller/test/sensor_tests/SensorTests.cpp:154: Failure
The difference between pressurePatient and patientPressures[index] is 1, which exceeds COMPARISON_TOLERANCE, where
pressurePatient evaluates to 0,
patientPressures[index] evaluates to 1, and
COMPARISON_TOLERANCE evaluates to 0.004999999888241291.
Patient Sensor at index6
controller/test/sensor_tests/SensorTests.cpp:154: Failure
The difference between pressurePatient and patientPressures[index] is 1.5, which exceeds COMPARISON_TOLERANCE, where
pressurePatient evaluates to 0,
patientPressures[index] evaluates to 1.5, and
COMPARISON_TOLERANCE evaluates to 0.004999999888241291.
Patient Sensor at index8
controller/test/sensor_tests/SensorTests.cpp:154: Failure
The difference between pressurePatient and patientPressures[index] is 2, which exceeds COMPARISON_TOLERANCE, where
pressurePatient evaluates to 0,
patientPressures[index] evaluates to 2, and
COMPARISON_TOLERANCE evaluates to 0.004999999888241291.
Patient Sensor at index10
controller/test/sensor_tests/SensorTests.cpp:154: Failure
The difference between pressurePatient and patientPressures[index] is 2.5, which exceeds COMPARISON_TOLERANCE, where
pressurePatient evaluates to 0,
patientPressures[index] evaluates to 2.5, and
COMPARISON_TOLERANCE evaluates to 0.004999999888241291.
Patient Sensor at index12
controller/test/sensor_tests/SensorTests.cpp:154: Failure
The difference between pressurePatient and patientPressures[index] is 3, which exceeds COMPARISON_TOLERANCE, where
pressurePatient evaluates to 0,
patientPressures[index] evaluates to 3, and
COMPARISON_TOLERANCE evaluates to 0.004999999888241291.
Patient Sensor at index14
controller/test/sensor_tests/SensorTests.cpp:154: Failure
The difference between pressurePatient and patientPressures[index] is 3.5, which exceeds COMPARISON_TOLERANCE, where
pressurePatient evaluates to 0,
patientPressures[index] evaluates to 3.5, and
COMPARISON_TOLERANCE evaluates to 0.004999999888241291.
Patient Sensor at index16
controller/test/sensor_tests/SensorTests.cpp:154: Failure
The difference between pressurePatient and patientPressures[index] is 3.9200000762939453, which exceeds COMPARISON_TOLERANCE, where
pressurePatient evaluates to 0,
patientPressures[index] evaluates to 3.9200000762939453, and
COMPARISON_TOLERANCE evaluates to 0.004999999888241291.
Patient Sensor at index18
[  FAILED  ] SensorTests.FullScaleReading (2 ms)
[----------] 1 test from SensorTests (2 ms total)

[----------] Global test environment tear-down
[==========] 1 test from 1 test suite ran. (2 ms total)
[  PASSED  ] 0 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] SensorTests.FullScaleReading
```

jkff, 2020-04-16:
I'm not sure I fully understood the intention of the test - is it perhaps only intended to be run on device? In that case, me converting it to googletest was premature and then please feel free to reassign back to me.

verityRF, 2020-04-17:
Oops! This was going to happen, this was just before we integrated pIO. I'll refactor it so it doesn't kill the test framework. Should be done by Saturday!

And no, you are correct, it (ArduinoSim) shouldn't be run on the (AVR) device.

closed 2020-04-20 by verityRF