RespiraWorks/Ventilator #117: Enforce consistent C++ naming conventions

jkff, 2020-04-17:
Currently our code uses a bunch of inconsistent naming conventions - there's a mixture of `like_this`, `likeThis`, `LikeThis` and `Like_this` for class/type names, functions, parameters, variables and constants.

Consistent naming would reduce the burden of choosing a naming style for new code people write.

https://github.com/mapbox/cncc is a tool that can enforce C++ naming conventions.

https://github.com/Project-OSRM/osrm-backend/blob/master/.cncc.style is an example config for it, within a large project that enforces its naming conventions by somehow running the tool in its `.travis.yml`.

martukas, 2020-04-17:
I'm personally partial to the google style guide: CamelCase for classes and like_this for variables, etc.. I'm guessing you might be familiar with this convention? :)

jkff, 2020-04-17:
Yeah, following [Google style guide](https://google.github.io/styleguide/cppguide.html#Naming) is a reasonable default choice and I am indeed familiar with it :)

For somebody who decides to take on this issue: it can be done incrementally e.g. in the following steps:

- Establish the enforcement infra but with an empty set of rules
- One by one add naming enforcements e.g. for files (forgot to mention that our files are also inconsistently named), functions, classes etc, coupled with fixes of that entity in the same commit

This would generate commits that are not quite as huge and tedious to author as one giant commit bringing all the code in compliance with everything.

Another approach, perhaps even better, is to first gradually bring things in compliance, and then add the tool enforcement + rules in the same commit with final fixups.

jkff, 2020-04-18:
I'm gonna try to take this one and enable clang-tidy while we're at it, as clang-tidy supports enforcement of naming conventions too.

martukas, 2020-07-29:
@jkff do you still intend to work on this or shall we untag you and keep this open for newbies or something?

jkff, 2020-07-29:
Yeah unfortunately I don't have time work on RespiraWorks these days - untagged myself. All the best to y'all!

martukas, 2020-09-11:
Just taking notes on the progress here.
Looks like you have to `sudo apt install clang-format`
I have tried this:
```
git clang-format --extensions cpp,hpp --style=google -v
```
but it seems to only fix whitespace and not variable naming conventions

dheater, 2021-02-06:
> Just taking notes on the progress here.
> Looks like you have to `sudo apt install clang-format`
> I have tried this:
> 
> ```
> git clang-format --extensions cpp,hpp --style=google -v
> ```
> 
> but it seems to only fix whitespace and not variable naming conventions

Apparently `clang-tidy` can enforce naming conventions. I've not used this feature yet, but the documentation is here: https://clang.llvm.org/extra/clang-tidy/checks/readability-identifier-naming.html

I don't see predefined style options (ex,`--style=google`) for `clang-tidy`.

asmodai27, 2021-04-17:
Looks like this is what we are looking for: https://gist.github.com/airglow923/1fa3bda42f2b193920d7f46ee8345e04

PR #512 (2020-06-14): [GUI] Keep Respiratory Rate range between 10-30 bpm.
Fixes #117
commit 1dfca8be (2020-06-14): [GUI] Keep Respiratory Rate range between 10-30 bpm.

Fixes #117