RespiraWorks/Ventilator #126: Implement sending ControllerRebootState message via nanopb

Miceuz, 2020-04-18:
**Summary:**
Add nanopb implementation of ControllerIdentification message that is being sent on reboot currently.

**Details:**
Add serializing functionality and use it in comms library.

jkff, 2020-04-29:
@jlebar Please add some details

jlebar, 2020-04-29:
@Miceuz can you clarify what you meant by this bug?  Is this "send software version from controller to the GUI"?

Miceuz, 2020-04-30:
Yes, now this issue is just about providing version information to GUI. 

Both firmware version and hardware version. I have asked electronics guys to include hardware identification via resistor combination on board, didn't yet look how they implemented that, but have seen the issue close.

martukas, 2021-05-05:
need solution for how to set hardware version, but the SW version part of this can be picked up and implemented by soemone

inceptionev, 2021-05-05:
Hardware Revision ID bits are on GPIO pins, described in the Rev A Interface Control Document lines 14 and 15: https://docs.google.com/spreadsheets/d/1EPEkAWZGx5ASWL2mgTzymS2f11Ow9ptx9c8Sycb3_Go/edit?usp=sharing

ID 00 is Rev 1.0, ID 01 is Rev A Hardware