RespiraWorks/Ventilator #163: Ensure the GUI can updating it's running docker container

hhff, 2020-04-25:


jkff, 2020-04-29:
@hhff can you elaborate? Is this a CI issue?

hhff, 2020-05-04:
Updated @jkff

jkff, 2020-05-07:
Hmm I don't think that self-update, or any possibility of network interaction at all, is a good idea for a medical device:

- If the current GUI is working fine for somebody, it's best to keep it that way rather than updating to a version that, who knows, might have a bug.
- It is especially risky to talk over the network given that if we don't get the security exactly right (which nobody does), someone might upload a malicious image.
- The GUI will likely be often used in places where there's no connectivity.

hhff, 2020-05-10:
I don't disagree, but in the absence of this workflow we'll need a manual way to very quickly get in touch with practitioners if we discover something unsafe about a specific version.

Most IOT devices I've worked on have a way to disable normal operation and nag for an update in the case of critical security updates.

In the absence of something like that, at least, we'll need to be sure @curtiskline and operations understand that's on them

hhff, 2020-05-10:
(Removed the milestone and made this Nice to Have)