RespiraWorks/Ventilator #164: Port Controller Units to CircleCI (test.sh)

hhff, 2020-04-25:


twslankard, 2020-04-26:
The CircleCI config for running controller tests is more or less in place, though the build fails at present (perhaps due to differences in tool versions?) I will investigate.

twslankard, 2020-04-26:
I've learned that CircleCI has a limited number of machine images, which do not seem to include a newer Ubuntu image. Not to worry, this should work within a Docker container. This is what I'll try next.

twslankard, 2020-04-26:
Okay, this approach works:

1. Build a Docker image based on Bionic that installs g++, git, python, pre-commit, and platformio.
2. Run the image in privileged mode an run `bash test.sh`. Privileged mode is necessary to enable ptrace, which is required by LeakSanitizer.

twslankard, 2020-04-26:
Qapla'!

Would love any suggestions or feedback you have at this time, @hhff, when it's convenient.

closed 2020-04-29 by twslankard