RespiraWorks/Ventilator #170: Ensuring blower is off during sensor calibration

verityRF, 2020-04-26:
Pressure sensor calibration (zeroing) can be performed, though in order for calibration to happen correctly, the differential pressure sensors must be experiencing the same pressure on both ports and the single ended pressure sensor must be experience ambient (atmospheric) pressure.

The way for these criteria to be meet is to have the blower off during calibration. 

As long as the sensors_init method is called before the blower is turned on in any form, calibration will be OK and subsequent readings will be OK. 

However, it is desireable for the system to routinely calibrate sensors by calling zero_sensors due to variations of temperature, age, and location over time. Thus ensuring the blower is off at these times is also crucial. 

The sensors.cpp tries to make sure the blowers are off but blowers_disable is an empty method. All blower access happens in place in pid.cpp instead of as a separate accessible module. 

Either: 
a) Pid.cpp ensures blower is off before calibrating sensors at any time
b) Blowers.cpp is implemented to provide universal access to blower settings

If neither of the above is met, fatal operation may occur in that sensor readings may be completely off due to an incorrect zeroization procedure.

jlebar, 2020-04-29:
@verityRF I attempted to update this bug based on the discussion we had a while ago; does this match your current thoughts?

tapemaster, 2020-07-22:
@inceptionev do you have ideas about calibration here?

tapemaster, 2020-07-22:
How to enter current altitude during calibration? Do we need an atmospheric sensor?

martukas, 2021-01-18:
related to #857