RespiraWorks/Ventilator #175: Setup Dockerhub for GUI Images

hhff, 2020-04-26:
We're going to be storing the base Raspbian image on Paperspace (@twslankard and @a-vinod are handling that with these tickets #2 #88 ), but that's going to be a lightweight image that simply runs the gui docker container.

In the future, we'll use `respiraworks/gui:latest` as the latest stable release, and our device will be able to pull this on a pre-established update event

hhff, 2020-05-07:
@twslankard - how i've done this in the past:
1. Make a generic Docker Hub credential for respiraworks, and add those credentials to our CircleCI Env Vars `$DOCKERHUB_USERNAME` and `$DOCKERHUB_PASSWORD`
2. Add a `gui-push-docker-image` job to the circleci spec that only runs on the `beta` and `release` branches
3. that job includes something like this (keeping in mind we can use the `docker_image_layer` cache feature of circleci to avoid double builds):
```
  DOCKER_IMAGE_VERSION="$(git rev-parse --short HEAD)"
  echo "$DOCKER_HUB_PASSWORD" | docker login -u "$DOCKER_HUB_USERNAME" --password-stdin

  docker build -f ../../Dockerfile ../../.\
    -t respiraworks/gui:$DOCKER_IMAGE_VERSION
```

Now, whenever someone rebases the repo on `beta` or `release`, and pushes, a new docker image will be cut like `respiraworks/gui:FDH3912` (where the end string is the SHA for that push)

These images can be public on our dockerhub, fwiw.

Nice to have: You can also alias that push (if it happened on the `release` branch) as `respiraworks/gui:latest` so that it's easy to find the latest stable build

twslankard, 2020-05-07:
Thanks for the assistance, @hhff. I'll start on this one ASAP. That release workflow sounds fine to me. I'll probably start with pushing `unstable` images with each successful build of `master` and then coordinate with the rest of the group regarding a release process that'll include the beta and release builds, to make sure we're all on the same page.

hhff, 2020-05-07:
Too easy. We can probably restrict pushing to those branches too

twslankard, 2020-05-14:
@hhff @twslankard @jkff 

Looking for advice, if y'all are willing. I'm running into a limitation. I'd like to be able to pass the GUI docker image between CircleCI jobs. (The docker push adds about a minute to the already 8m long gui-build-image job.)

Logically, I should use the `setup_remote_docker` option, though this doesn't support the version of Docker we need.

I tried using a shared workspace, but the executor doesn't have permission to access `/var/lib/docker`.

A kludge that I haven't tried yet would be to use the workspace in conjunction with `docker save` and `docker load`.

Any other ideas or suggestions?

jlebar, 2020-05-14:
> Idea 2: [...] I feel like this is a weird approach because it won't be apparent in the CircleCI workflow gui whether an image has been pushed.

Honestly I am fine with that.  It'll still be apparent in the logs.

Another option would be to use CircleCI's caching mechanism.  Presumably you only care about files in gui/ and common/ for the GUI build, so there are going to be lots of cache hits normally.  If a cache hit were fast enough, the "push image" step could "rebuild" the whole image, just with a guarantee that it's a cache hit.

Another another option would be to have two jobs, one which builds and pushes, and one which just builds.  Run build-and-push on master, run push-only on non-master.

twslankard, 2020-05-14:
Thanks for the input, Justin. I'll think I'll start with the conditional step and see if that works.

closed 2020-05-14 by twslankard