RespiraWorks/Ventilator #208: Determine demo device for alpha

jkff, 2020-04-29:
TODO(@jkff): follow up with Edwin/Ethan and fill this in

Please take the time to describe the issue in detail, so that other contributors can pick it up and implement or declare it done/obsolete in case if was completed out of band.

**What has to be done**
Please describe what has to be done, ideally in declarative form - what does success look like.

**How do you know it has to be done**
Please link any documents, Slack threads, quote personal communication, reference your personal domain expertise etc. explaining why you think this should be done.

**Code pointers**
Please give some thoughts on roughly where in code do you think the necessary changes might be located.

martukas, 2020-05-06:
Since the demo is done, I guess this is now obsolete and can be closed?

jkff, 2020-05-07:
Yup.

closed 2020-05-07 by jkff