RespiraWorks/Ventilator #224: Pressure setpoint square wave is not always consistent when running on Arduino

jlebar, 2020-05-01:
Follow the instructions below to chart a pressure setpoint graph from the Arduino.  This shows the system pressure the Arduino *wants* to achieve.

This chart should be a square wave with a consistent period.  But every once in a while, we "miss a beat" and the inhale or exhale is too short or long.  It's very obvious in the graph.  It happens periodically, and for me always happens first around t=300 in the graph:

[image]

You don't need pressure sensors hooked up or anything; I've reproduced it with the Arduino alone.

- Build at d8020002bc43fc794bf59613c79cb506104455a0, running on an Arduino RF Nano.

- Enable NO_GUI_DEV_MODE by uncommenting this line https://github.com/RespiraWorks/VentilatorSoftware/blob/d8020002bc43fc794bf59613c79cb506104455a0/controller/src/main.cpp#L68

- Run on the device, pio run -t upload

- Immediately start capturing a pressure trace using the Arduino IDE (tools -> serial plotter).  Use 115200 baud.

**What is wrong**
Square wave should be consistent, but it's not; it sometimes exhales for too long (attached image above).

**How do you know it is wrong**
By definition command-pressure has a consistent pressure wave.

**Code pointers**
`blower_fsm.cpp`?

jlebar, 2020-05-01:
I don't know what's happening here.  It's tempting to think that this is due to some operation taking way too long and hanging up the whole main loop.  But I don't think that's it; if it were, we wouldn't be receiving *any* data from the device.

Something else is wrong.

jlebar, 2020-05-01:
Putting this in the alpha controller milestone because I think it would be pretty embarrassing for us to show a graph to UCSF or whoever if we can't even consistently breathe in and out with the right timings.

martukas, 2020-05-03:
What are the units on the x-axis? How bad is the delay?

jlebar, 2020-05-03:
One tick on the x-axis is 1/10s.  The square wave is one breath.  This is a highly significant delay.

Miceuz, 2020-05-03:
I can't reproduce it on STM32. Might be related to some long-ish float calculation.

jlebar, 2020-05-03:
Thanks for running that test, Albertas.  That's really useful.

> Might be related to some long-ish float calculation.

If the calculation were stuck, the whole main loop would be stuck, and we wouldn't receive any data for that period of time.  But we *do* receive  data, indicating that  the loop isn't stuck.

jlebar, 2020-05-03:
Something is wrong with the timer.

In this chart, the orange line that's a step function is charting `Hal.now().millisSinceStartup() / 1000 % 10`.

[image]

This next chart tracks both native Arduino `millis()` and `Hal.now().millisSinceStartup()`.  I am not sure which line is which, but I think it's the Hal line that's misbehaving.

[image]

jlebar, 2020-05-03:
Ah, this is 16-bit overflow in the hal timer.

commit d1fffe99 (2020-05-03): Fix 16-bit overflow in Hal timer code.

`int` on Arduino is 16 bits wide, and 2^15ms is only about 30 seconds.
Therefore a signed int overflows quite frequently!

Fixed to be int64_t, as it should be.

Fixes #224.

GPC: timer-overflow
commit cefd12d4 (2020-05-03): Fix 16-bit overflow in Hal timer code.

`int` on Arduino is 16 bits wide, and 2^15ms is only about 30 seconds.
Therefore a signed int overflows quite frequently!

Fixed to be int64_t, as it should be.

Fixes #224.

GPC: timer-overflow
commit e87b64e0 (2020-05-03): Fix 16-bit overflow in Hal timer code.

`int` on Arduino is 16 bits wide, and 2^15ms is only about 30 seconds.
Therefore a signed int overflows quite frequently!

Fixed to be int64_t, as it should be.

Fixes #224.

GPC: timer-overflow
PR #231 (2020-05-03): Fix 16-bit overflow in Hal timer code.
<git-pr-chain>

#### Commits in this PR
1. Fix 16-bit overflow in Hal timer code.
    
    `int` on Arduino is 16 bits wide, and 2^15ms is only about 30 seconds.
    Therefore a signed int overflows quite frequently!
    
    Fixed to be int64_t, as it should be.
    
    Fixes #224.

#### [PR chain](https://github.com/jlebar/git-pr-chain)
1. 👉 #231 Fix 16-bit overflow in Hal timer code. 👈 **YOU ARE HERE**
1. #232 First pass at tidal volume measurement.
1. #233 Re-tune PID.
1. #234 Change default params in DEV_MODE_comms_handler.
1. #205 Clarify comment o [... cut]
closed 2020-05-03 by jlebar