RespiraWorks/Ventilator #239: PID code should use float, not double

sglow, 2020-05-03:
At the moment the PID routines use double precision floating point numbers.  That's unnecessary and too slow.  These should be replaced with single precision (32-bit) floats.

**What is wrong**
The Arduino being used in the alpha build doesn't support doubles at all, declaring a double just gives you a 32-bit float.  Declaring the data type as double doesn't do anything but cause confusion because it implies that doubles are being used when they really aren't.
On the STM32 which will be used for the beta and later builds, 32-bit floating point math is supported in hardware and is very fast.  64-bit double precision floating point math is also supported, but will be implemented by much slower software routines.  Orders of magnitude slower.

**How do you know it is wrong**
None of the math we're doing requires the precision that double precision floating point will give, but speed is important.
Doubles should only be used where there's a justifiable need for the extra precision and that need out weights the cost of using them. 

**Code pointers**
This is actually a really simple fix, but grep for double and replace with float anywhere in the code.  I'm going to mark it as a good first issue.

sglow, 2020-05-03:
Actually, good first issue doesn't seem to exist anymore.
Anyway, this is simple.  I'll fix it this week if no one beats me to it.

asmodai27, 2020-05-03:
Will treat this after https://github.com/RespiraWorks/VentilatorSoftware/pull/221 merge ;)

sglow, 2020-05-03:
Perfect, thanks.

closed 2020-05-05 by asmodai27