RespiraWorks/Ventilator #254: Instructions for building for and deploying to STM32

martukas, 2020-05-06:
**What is wrong**
There are no instructions for building for and deploying to STM32. 

**How do you know it is wrong**
In slack we have had anecdotal reports of people managing to run it on STM32, but how this was done has not been documented.

**Code pointers**
@Miceuz @sglow I believe you had success here. Could you please share your progress?

sglow, 2020-05-06:
Building for the STM32 is easy thanks to the nice platformio setup that someone (@jlebar possibly?) set up.  You can build for the nucleo board using the Arduino libraries using:
pio run -e nucleo

I've got an alternate build which replaces the large Arduino libraries with a minimal HAL layer that writes directly to the hardware.  This is what all my testing has been on so far.   Once that gets merged you will be able to build that version using:
pio run -e stm32

Either way, you can use the visual studio code debugger to load it to the nucleo board and run.  I was pleasantly surprised that this MS product worked well even on a Linux host.    

What you can actually do with the nucleo board is fairly limited at the moment due to the minimal hardware, but in another week or so we should be able to plug the nucleo board into the mounting board that Edwin designed and actually do something useful with it.

jlebar, 2020-05-06:
You should also be able to do `pio run -e stm32 -t upload` to build and
upload in one step.

martukas, 2020-06-26:
I believe we now have adequate instructions on how to deploy. There are additional nuances on how to deploy when you have multiple controllers all attached to the machine, but that is also covered under in `integration_testing` docs.

closed 2020-06-26 by martukas