RespiraWorks/Ventilator #255: GUI CPU use too high

twslankard, 2020-05-06:
**What is wrong**
Note, I still need to confirm this bug. Using `top`, I observed a very high CPU use ~ 250% while running the Qt GUI in a Docker container on the RPi. I was also connected to the RPi using SSH and VNC. I also observed a significant lag in my SSH terminal while the container was running. This issue was not present before the container was running, and it ceased after I terminated the container. (Okay, it's the container.)

I hear that we are getting support from another group to develop a new GUI, so I'm not certain if this issue is a priority. What do you think, @martukas ?

martukas, 2020-05-06:
Is it actually 250%? Or is that just some fluke of how the VM reports stuff? If it doesn't melt the hardware, I'd say it's not super urgent :)

twslankard, 2020-05-06:
This is running an an actual Raspberry Pi, which becomes pretty hot and nearly unresponsive while the GUI is running. My guess is either Docker is introducing too much overhead (which I doubt, but it's possible) or there's an issue with the GUI performance.

jkff, 2020-05-06:
The GUI is not very optimal. It communicates with the controller [every 30ms](https://github.com/RespiraWorks/VentilatorSoftware/blob/35468850cd2f1bb948043388a3a3e7308ad725dc/gui/main.cpp#L84), and re-renders the graph [every 60ms or so](https://github.com/RespiraWorks/VentilatorSoftware/blob/35468850cd2f1bb948043388a3a3e7308ad725dc/gui/main.qml#L14) . At least the latter frequency could be reduced. There's also a fair amount of copying C++ objects going on, I reduced some but maybe there's more.

I don't know whether or not it is necessary to optimize this. Assuming it is: is it possible to run a profiler on the pi and see what's actually the bottleneck?

twslankard, 2020-05-06:
@jkff good call. I did spend a little time attempting to use `perf` but ran into an issue where it is unavailable in the version of Raspbian I'm using. (I'll continue to work on this, but I think I'll have to download and install the source.) Let me know if you have other ideas I should try.

jkff, 2020-05-07:
@twslankard Try changing the graph re-rendering timer to be less frequent? E.g. if you set it to be once per second, does that get rid of the high CPU usage? If yes, then the issue is there :)

Another fix we can make is, in the `communicate` `PeriodicClosure` - we have to do a round of communication every 30ms, but we don't have to store the result of _each_ of these rounds into `ControllerStateHistory`. We could e.g. store only one every 100ms. But would be good first to confirm that this is the issue.

twslankard, 2020-05-28:
Bumping this to High Priority. Right now, the RPi overheats after a few minutes.

twslankard, 2020-05-28:
@jkff it does appear that the GUI updates affect the CPU usage significantly, though not completely. Here's what I've tried so far:

* Changed the graph update frequency to 1hz. CPU usage was still approximately 250%, and the RPi overheated within 5 minutes.
* Changed the graph update and receive callback frequency to 1Hz. GPU usage was approximately 70%. This still seems high to me for an app that isn't doing much yet.

I'm not certain how conclusive this is, though it does appear GUI rendering is one potential bottleneck at present. I'm going to try to learn how to use the QML profiling tools and see if there's anything going on there, though I may also try to see if running outside of Docker helps too.

jlebar, 2020-05-28:
If you have a Linux box, I expect this will immediately yield to `perf
report`.  You can also run that directly on the Pi.

twslankard, 2020-05-29:
I may be blocked on this issue (I've discovered by SD card is too small to permit much development on directly on the RPi.) If anyone is willing to help run perf against the GUI on the device, I'd be very grateful.

In the meantime, I'm going to attempt profiling on my Linux box.

paulovap, 2020-05-29:
Hi folks, I was working on a hackathon project at my company (we built a mobile device farm cabinet using rPi) and faced a similar issue. Not sure if exactly the same but I think it is worth mentioning. I will just paste on I had in the documentation. When I get my hands on a Rapsberry PI I will help you in the investigation.

> We developed the "waiting" screen using [Qt](https://www.qt.io/). But, when it ran on the PI CPU usage went through the roof and the animation on screen was very poor.
> 
> We notice an error message `warning: DRI2: failed to authenticate` which pretty much means there is no graphic acceleration going on. After some debugging, we realized that `libEGL` that comes with Linux distro was not working properly.
>
> The obvious question that came to my mind was: "How people play Minecraft in this thing?". After checking minecraft-pi distribution, it turns out that Minecraft ships its own `libEGL` implementation. So, we created a symbolic link for our app, to use that one, and VOILA! Smooth 60fps with very low CPU usage.
>
> So now our cabinet depends on `minecraft-pi` package. If this is not a hack, I don't know what a hack is. :)

twslankard, 2020-05-29:
@paulovap that is amazing. I'll try it. Thanks for the info!

paulovap, 2020-05-29:
@twslankard seems like in forum people had similar issues:

https://raspberrypi.stackexchange.com/questions/61078/qt-applications-dont-work-due-to-libegl

twslankard, 2020-05-29:
I've tried a couple more things, but so far I've had no luck.

* I installed the `minecraft-pi` package just for fun. Although I couldn't prove it, it appeared that graphics acceleration was working just fine.
* I created symbolic links for libGLES and libEGL in `/usr/lib/arm-linux-gnueabihf/` and ran `sudo ldconfig` to reload the shared lib cache. (I still get `DRI2: failed to authenticate`.) I cleared all docker image and containers and re-pulled and re-ran, but no luck.
* I installed all Qt deps directly on the RPi (running Raspbian) and built the GUI on the device, running it w/o Docker. Although I experienced no linking issues, I still encountered `DRI2: failed to authenticate`. 

My next step will be to build Qt5 from source to see if that helps.

paulovap, 2020-05-29:
Folks, I manage to install the latest Raspberry OS on my Raspberry PI 4 and the project run smoothly "out-of-the-box".

I am quoting because I was messing with `raspi-config` option before, like:
* Increase GPU memory to `128mb`.
* Change Compositor
* Change GL driver

Not sure if any of this had any effect.

Steps:

1) I created a file `kms.json` to force Qt to run on eglfs in the mode I wanted
```json
{
  "device": "/dev/dri/card1",
  "hwcursor": false,
  "pbuffers": true,
  "outputs": [
    {
      "name": "HDMI1",
      "mode": "1920x1080"
    }
  ]
}
```
Probably the modes will change depending on the Screen, but you can try different modes.

2).

I wrapped this script to get more information about graphics on Qt5. I called it run_eglfs.sh
```bash
#!/bin/bash

#Provides Rendering debug information
QT_LOGGING_RULES=qt.qpa.*=true

DISPLAY=:0

QT_QPA_EGLFS_KMS_CONFIG=kms.json QT_QPA_EGLFS_INTEGRATION=eglfs_kms build/ProjectVentilatorGUI -platform eglfs
```

Unfortunately, I don't have permission to push on this repo yet. Once I have I can add this script to make it easier.

@twslankard let me know if you still facing issues and the output of your run.

jlebar, 2020-05-29:
> I don't have permission to push on this repo yet

Invite sent!

paulovap, 2020-05-30:
@twslankard try https://github.com/RespiraWorks/VentilatorSoftware/pull/399 and paste what is the output of the run.

martukas, 2020-08-05:
We should log CPU usage always (either in debug mode while developing, or maybe even always).

martukas, 2021-05-05:
The VM aspect of this is no longer relevant, but we should still log and display CPU usage in GUI.

martukas, 2021-05-05:
GUI could also log controller CPU usage to file if it is conveyed via pbuffer