RespiraWorks/Ventilator #257: PID loop should be synchronous

sglow, 2020-05-06:
Time matters in a control system.  The proper functioning of a PID loop is very much dependent on it running at a specific and consistent update rate.  That's not currently happening in our system.

**What has to be done**
Presently, the controller runs a single main loop which includes things like handling the communications interface with the GUI and performing the PID loop calculations.  It's likely that more will be added to this loop in the future.  This means that the frequency at which the PID calculations are being executed is not well defined and could change depending on how long other functions in that loop take.

The PID library handles this situation to some extent by checking the time since it was last run and only performing calculations if more then a preset sample time has elapsed since it last performed the calculations.  This helps, but can only handle cases where the PID loop is being called too frequently, not too slowly.  Also, this approach leads to jitter in the system dependent on the granularity of the timer being used (1ms, not very fine) and the timing of the other functions in the loop.  This approach is better then nothing, but a poor substitute to a properly synchronous loop.

One possible solution would be to add an RTOS to the system, but I don't advocate this.  This would add undue complexity and isn't necessary.

A better solution would be to run the less time critical functions (such as the communications interface) in the main loop as they currently are, and run the time critical sections from the code from a timer interrupt which is triggered by the HAL at a specific frequency.  This high priority update should include sampling of the A/D inputs, calculating the system set points, performing the PID calculations and updating the PWM outputs.  

**How do you know it has to be done**
The I and D parameters in a PID loop both have a timing component to them.  The I term is based on the Integral of the error which is calculated by summing the error each time through the loop.  The resulting integral is dependent both on the error that's added and the frequency which which it's updated.  The D term is based on the Derivative of the error, i.e. the amount the error has changed over time.  The gain that this term provides is directly related to the time at which it's updated.

I don't think that saying that timing is important in PID loop calculations is controversial, but can provide more background material if needed.

**Code pointers**
A new feature will need to be added to the HAL.  It will require an interrupt to be generated from a timer at a fixed frequency.  The interrupt handler in the HAL will call a system function which will perform the loop updates.  That function should probably be added to the main.cpp file where the lower priority loop also resides.

The frequency of this timer update should probably be at least 1kHz.  The current loop update rate of 100Hz is likely too slow for us to get good results.  Running the main loop at 1kHz will not be a problem for the STM32 processor (I use a similar processor on servo drives which run much more complex loops at 16kHz), but may be too much for the poor Arduino used in the alpha system.  For that reason I'm going to mark this issue as an upgrade for the beta controller.

sglow, 2020-05-06:
I'd be happy to work on this, but am not sure the Arduino can support it.  Perhaps it's time we forked the project to start working on beta controller specific features?  The alpha hardware is important, but I think it's blocking progress on the arguably more important beta.

asmodai27, 2020-05-06:
> This approach is better then nothing, but a poor substitute to a properly synchronous loop.

Agreed, this is good for a proof of concept, but if we were to actually run this in more demanding conditions, this design's limits will probably become an issue.

>  The current loop update rate of 100Hz is likely too slow for us to get good results

From what I gathered about our blower, and especially the fact that it has a lot of inertia, I'm not sure this is actually an issue while we keep the pneumatics' design of Alpha.
But I hear we are adding a valve to allow the blower to stay ON most of  the time... this means the controller will have to adapt to the new actuation system, and we will have to change the current PID design anyway when that time comes.

closed 2020-05-27 by sglow