RespiraWorks/Ventilator #263: Extract the purely computational part of the controller

jkff, 2020-05-08:
**What has to be done**
We would like to test the controller using a simulated physical device in Modelica - which will simulate all of the ventilator and patient, and will provide simulated sensor readings to the controller and simulate act on its outputs.

In order to do that, the controller must contain a purely computational component that can be used in Modelica - essentially a C++ class of the form:

```
struct ControllerOutputs {
  float blower_speed;
  bool exhaust_open;
};

class Controller {
  // Performs 1 round of control.
  ControllerOutputs Run(Time now, VentParams params, SensorReadings readings);
};
```

(naming is not precise, but this is the basic idea)

Currently this logic is [here](https://github.com/RespiraWorks/VentilatorSoftware/blob/76ed2ddfe44a96ebde48d0a83a8600bdd27d5dd9/controller/src/main.cpp#L121-L127) and contains reading sensors and writing pins inside it (specifically inside `blower_pid_execute`, I think).

**How do you know it has to be done**
Several [Software Team meeting notes](https://docs.google.com/document/d/14lrTVcYuuUii7QTvuJJneex-_uKSHQ33Hg9CR5pOARo/edit#) mention the need for simulated integration testing; see specifically notes talking about Modelica and Modelon.

Specifically, we want to take Modelon's existing model of the ventilator, and replace their Controller component with one running our controller code, and then simulate the resulting model for testing purposes in a variety of scenarios.

Modelica has an External Object interface, and the Controller must fulfill that interface. John Batteh is working with us to give us a model into which the C++ controller can be plugged, but we must refactor it into a pluggable form.

**Code pointers**
https://github.com/RespiraWorks/VentilatorSoftware/blob/76ed2ddfe44a96ebde48d0a83a8600bdd27d5dd9/controller/src/main.cpp#L121-L127 is the code that has to be refactored.

asmodai27, 2020-05-08:
Just did a first step towards this in branch jtravert/issue263_refactor_computation by removing the pin references from blower_pid.
Will try to finish tomorrow with build of the actual class.

jkff, 2020-05-08:
Fantastic, thanks @asmodai27 !

asmodai27, 2020-05-09:
I guess https://github.com/RespiraWorks/VentilatorSoftware/pull/259 will be ready soon, and I will rebase to this branch in order to pass now to the PID and replace all Hal.now() references to now...

jkff, 2020-05-10:
A bit more work remains that I'm doing in a pending PR.

jkff, 2020-05-11:
After #274 the only remaining thing will be:

Ideally we should move the sensor calibration code into Controller too, so that we are able to simulate how it copes with (simulated) noise.

jkff, 2020-05-11:
On second thought we actually shouldn't move sensor calibration code into Controller, at least not for the purpose of integrating with Modelica: Modelica will invoke the controller once per simulation step, so it won't be able to get multiple sensor readings. With that in mind, this issue is fixed to the extent it can be.

closed 2020-05-11 by jkff