RespiraWorks/Ventilator #300: Control of Stepper Motor for Inhale Proportional Valve and Exhale Proportional Valve: Pinch Valve

bruprescott, 2020-05-13:
Implementation of valve control in the software:
Where can I find the control files for stepper motors?

The valve (Pinch Valve) is mentioned in the diagram below:
https://github.com/RespiraWorks/PhysicalDesign/blob/master/Alpha-BOM/pneumatics-pizza.png

sglow, 2020-05-13:
It's not there yet, but I'm working on adding support for the motor controller to our HAL.  Should have that by the end of the weekend at the latest.

jkff, 2020-05-13:
@bruprescott Could you rephrase the issue? Is this something that has to be implemented, or just a question about where certain code is?

sglow, 2020-05-13:
I may have misunderstood the question.  What I'm working on is the low level interface to the stepper motor driver that we're planning to use for testing out the various pinch valves.  I started work on it last night, but diverted today to do some work on the debug interface.  I'm hoping to have both finished this week.

bruprescott, 2020-05-14:
@jkff 
This was a question about where I could find the code for implementing the stepper motor control described in the diagram.

I´m also developing a respirator with proportional valves using stepper motors. A mixture of ideas, at the beginning of the MAK-AIR staff.

I use L6208PD controller and motors with 45Kgf / cm.

jkff, 2020-05-14:
@bruprescott Got it. However, Github issues are for tracking bugs or new features so I'm going to close this one - please use Slack for questions. For this one the most appropriate channel is #software-controller.

closed 2020-05-14 by jkff

sglow, 2020-05-14:
I just took a quick look at the data sheet for the L6208PD driver.  I don't think the code that we end up creating for our stepper driver will be very helpful.  We're using the powerSTEP01 part which has a built in controller with an SPI interface, so we'll be interfacing to it at that level.

ST has a few other chips with a similar high level SPI interface that might be of interest to you, the L6470 has a similar high level SPI interface but comes in an easier to solder package then the powerSTEP01 chip.  I don't think it can handle as much current though.

I didn't spent too much time with the L6208 data sheet, but it looks like it's expecting a pulse train input that determines how quickly it steps the motor.  The chip basically gives you the back end power electronics for pushing current to the motor, but expects you to handle trajectory generation (i.e. increasing the rate of steps to come up to speed, and reducing them to slow to a stop).  That higher level function is built into the parts we're looking at.

If you end up using a part similar to ours, please feel free to take anything from our code that you find helpful.

Best regards & good luck with your project.
Steve

bruprescott, 2020-05-14:
I looked at powerSTEP01 and L6470. Both can be used. I will migrate my hardware to use it.
PowerSTEP is far superior and at a similar cost. I believe it is the best choice. I will follow in your footsteps.
Welding it is not a problem, I have a complete laboratory.

Thanks for sharing!

bruprescott, 2020-05-18:
@sglow 

Hello,

I already have hardware ready to test the proportional valves.

Do you plan to implement the control code for powerstep01?

What is the strategy to control the three valves (inhale, exhale, oxygen) in the software?

Will oxygen have a pressure regulating valve at the outlet in the cylinder?

sglow, 2020-05-18:
I'm working on control of the powerstep01 right now.  You can take a look at the steve_working branch if you're interested.
Right now I'm just working on low level interface to the chip.  I haven't thought too much about higher level questions yet.