RespiraWorks/Ventilator #307: Need a design for self-test mode

jkff, 2020-05-14:
Per the system requirements (which as of writing can be found [here](https://github.com/RespiraWorks/VentilatorSoftware/wiki/Beta-Milestone#functional-requirements) - see [System requirements](https://drive.google.com/file/d/1HgBBUELFsKxpqV9qQ_b9u50Z6b0gqHHX/view), requirement 131 "Pretest mode"), we must have a pretest mode:

_The ventilator shall have a pre-test mode where the inlet and outlet ports are connected and the  operation of the device is checked including, capability to product correct pressure, function of valves, function of tidal volume measurement._

The existence of this mode raises many questions:
* Should it run on every start-up, or on-demand, or both? 
* Should the device refuse to function until a self-test?
* Should it function only if self-test passes, or can you ignore some errors?
* Should it confirm that you don't have the patient hooked up?
* etc.

**What has to be done**
We need a design for this mode:
* How does it fit into the overall UX of the ventilator
* How is it implemented at a protocol level - i.e. what does the GUI say to the controller, what does the controller say to the GUI.
* It is explicitly OUT OF SCOPE of this design to define which specific self-tests are to be performed. This is just about the protocol and UX; the specific tests will be the subject of a separate issue.

Success criterion:
* the design is written up on wiki (or in the repo) in the form of a Markdown document
* a reasonable person could read the design and confirm that it is in principle capable of supporting at least the tests described in the system requirement
* a good coder could read the design and start implementing it

**How do you know it has to be done**
System requirements say so (see above).

**Code pointers**
As of writing, just `network_protocol.proto` - the protocol will have to be an extension of that.

jkff, 2020-05-24:
Note: This is part of https://github.com/RespiraWorks/VentilatorSoftware/wiki/UI-Requirements.

martukas, 2020-08-07:
Here is an outline posted on May 13th by @inceptionev in #self-test channel (about to be archived around Aug 7)
[original post](https://respiraworks.slack.com/archives/C0135M1BHHD/p1589403525001800)

This is the stub we wrote for Code Life earlier:
1.0 User Procedure
1.1   Power-On Self-Test
At power-on the unit will verify presence and communication between all its internal components, and verify nominal quiescent readings on all its sensors.
1.2   Automated Validation Self-Test
Note: for safety, the Self-Test feature is not available once any patient ventilation program has been selected on the ventilator. The Self-Test is only available after a power-cycle of the unit. The user will tap the Self-Test button on the main screen of the ventilator, which will then prompt the user to connect a standard ventilator hose directly from the inhalation port to the exhalation port. Once the user has confirmed, the unit will automatically proceed with two automated test sequences:
1.2.1 Closed-flow test
For this test, the unit closes the exhaust valve and validates the following:
* Proper blower function
* Proper pressure sensor function
* Respiratory cycle control and timing (PIP, PEEP, RR, I:E)
* Proper exhaust valve function
* No system leaks
* Proper FiO2 function
* Presence and function of the humidification chamber
1.2.2 Open-Flow test
For this test, the unit opens the exhaust valve and validates the following:
* Proper flow sensor function
* Open-circuit detection and alarm
* Proper volume control function (TV)