RespiraWorks/Ventilator #319: PID loop needs work

sglow, 2020-05-14:
There are several problems with the PID calculations

**What is wrong**

- The loop really needs to be synchronous, that's already listed as issue #257. 
- Right now we're using a separate PID library for the calculations.  This makes it much more difficult to follow what's being calculated then it really should be.  The basic PID calculations are extremely simple and don't need to be buried in a library.
- The D term in the PID is not currently being calculated in a standard way.  The D term should be multiplied by the change in error, not change in input.  It's debatable whether we need a D term at all, but if we are going to have one it should be properly calculated.
- I don't think the clipping is working as it's intended.  Based on the code I think we expect the output of the PID library to be in the range 0 to 256 and the output of the controller to be in the range 0 to 1.  It's not, the PID library is only clipping the integral term.  P and D terms are added after that clipping.
- The use of the PID library makes this inflexible.  Our system is getting more complex as we add both valves and blower to control.  We need to be able to update the calculations is a more readable way.

**How do you know it is wrong**
Code examination.

jkff, 2020-05-15:
Per https://github.com/RespiraWorks/VentilatorSoftware/pull/321#issuecomment-629013303 I disagree with bullet 2.

For bullet 3 - can you elaborate? I believe we support both modes; the logic you're talking about is [here](https://github.com/RespiraWorks/VentilatorSoftware/blob/d2c671a531df0c5f967ace05101d55aae0246016/controller/lib/pid/pid.cpp#L76).

Not sure I understand point 4 too.

asmodai27, 2020-05-15:
Regarding point 3, there was a discussion of this when I first worked on the PID, available on slack: 
https://respiraworks.slack.com/archives/C011CJQV4Q7/p1588361829371300?thread_ts=1588361829.371300&cid=C011CJQV4Q7

Bottom line: this is a bit less fast to respond to changes in setpoint, but mitigates the risk of oveshooting, which is critical in our case.

sglow, 2020-05-15:
The PID library does have an option to treat the D term like a D term, we just aren't currently using it.
If you set DifferentialTerm::ON_MEASUREMENT to true then you don't have a D term at all, you have a feed forward term.  I have two major issues with this:
- First, the `DifferentialTerm::ON_MEASUREMENT` setting is far from intuitive as to what it does.  There's zero documentation of what that setting means, at least none that I can find.
- Second, using the kd gain for feedforward when some obscure setting is enabled is a very poor approach.  Feed forward terms are useful, and a good PID library might include them, but it shouldn't be an either/or with the differential term and it certainly shouldn't quietly use the D gain if the setting is enabled.

closed 2020-06-24 by sglow