RespiraWorks/Ventilator #355: SiL (this should be Epic)

ysz, 2020-05-24:
We want to run [Modelon](https://www.modelon.com) "model's of the ventilator in Modelica" , [and model of the patient too] https://gitlab.com/Modelon/PandemicVentilator against [Brain build](https://github.com/RespiraWorks/SystemDesign/blob/master/4_Manufacturing_Design/Alpha_Build_Instructions/brain_build.md)

In the end, we want to build SiL for the latest code of everything including GUI from our repositories to be tested on CI/CD

This issue WIP tracks progress.

We built an example and @jkff  was testing it out and, John Batteh thinks, was ready to connect code to be written here.

**What has to be done** 

Sub-tasks?

1. Talk to Elizabeth and John Batteh  
1. (Have pizza build)
1. Make Modelon model run our controller code (TODO: code pointer) on that Windows VM ...
1. Maybe try Platformio? @jkff's 

**Code pointers**

Get access to VM with Modelon (ask @curtiskline )

- VM has solution named "Visual Studio" where "controller" is replaced with C code as per @jkff 

    See custom build step there.

- https://github.com/RespiraWorks/VentilatorSoftware/blob/master/controller/lib/core/controller.h

**Other**

### Modelon model of teh patient 

We can change different variables there. ( Resistance, compliance, min volume, air temp in body) They can enable time-traced changes to these variables. [RespiraWorks Software meeting notes](https://docs.google.com/document/d/14lrTVcYuuUii7QTvuJJneex-_uKSHQ33Hg9CR5pOARo/edit#)

ysz, 2020-05-24:
Update

- I've got Paperscape access to Modelon VM

ysz, 2020-05-26:
Update
- I've received Modelon license from John

jlebar, 2020-06-03:
Sorry, what's SiL?

martukas, 2020-06-04:
Software in the Loop. Which is what the whole modelica thing is.