RespiraWorks/Ventilator #358: Add unit tests for sprintf.h

jkff, 2020-05-24:
**What has to be done**
We have our own sprintf authored by @sglow in [sprintf.h](https://github.com/RespiraWorks/VentilatorSoftware/blob/7131e439303234f689ba33426e6157a902ffa569/controller/lib/debug/sprintf.h) and cpp. The reason we rolled our own instead of the standard one is that we want it to be malloc-free.

This is several hundred lines of complex string formatting code and it doesn't have unit tests.

**How do you know it has to be done**
Without extensive unit tests, the library is not safe to include in the product. String formatting is notorious for producing buffer overflow bugs. Extensive unit testing is one of the ways this risk could be reduced.

We of course have the option of just not including the library in production builds at all, but then we lose the ability to debug something happening in a production build.

**Code pointers**
Add unit tests in a new file `controller/lib/debug/sprintf_test.cpp`.
I think we should, at least, borrow [this test suite](https://github.com/mpaland/printf) (if not borrow the whole library - which appears to be of comparable complexity to ours).

CC @jlebar

martukas, 2020-08-29:
The code is badly out of date with modern compilers. Any use of `long`, `long long` and so on should be replaced with std types such as `int32_t` and `int64_t`. As it is now, it is not passing clang-tidy checks.

martukas, 2020-08-29:
Actually, I second @jkff 's suggestion that this be replaced with the suggested library.

martukas, 2020-09-13:
The [mpaland/printf](https://github.com/mpaland/printf) library appears to work, but needed some modifications, because it is geared at C++11, for example some type conversions are implicit.

Instead of committing the proof-of-concept code, I have gone forward with forking the original library as:
https://github.com/RespiraWorks/printf

We will update it there and ensure it is well tested. It will also be easier to maintain separately in case the original library gets updated and we need to rebase. Once this fork is in reasonably good shape, we will import the library into Ventilator as either a git submodule or just copy a specific snapshot.

martukas, 2021-02-15:
Looks like sprintf is not actually being used by in the controller. It appears that it was there only for printing out traces. The debug interface has a mode for dumping this "trace buffer" over serial, but there is nothing that actually fills it up. Since it is not actually used and no one has actually felt inspired to use it recently, we propose to actually drop this from our code base. We may leave a stub pointing to the `mpaland` library noted above, so that in case someone does need it they are welcome to go and bring in this well-tested code.

asmodai27, 2021-03-03:
sprintf has been removed by #1064, since we no longer use debug.Print anywhere in the code.
In case we want to resurrect it, codebase after #1063 has it in a form that is compatible with the current debug library architecture.