RespiraWorks/Ventilator #366: Implement pressure alarm

tapemaster, 2020-05-26:
#SR-20

Alarm shall trigger when inspiratory airway pressure exceeded limits:

Pplat <30-35 cmH2O

Peak P no more than 2 cmH2O greater than Pplat

Fail-safe valve opens at 60cmH2O (powered or un-powered)

jkff, 2020-05-31:
[These notes](https://docs.google.com/document/d/1QFFcg2gL3PXXzyqgSBz3XZQm_oobZ2osFuErh492UEA/edit) have some guidance on setting high pressure alarms: "high pressures (when pts start to cough/become uncomfortable, set 20cm above peak pressure)"

dtgerson, 2020-06-06:
Note: because of how we're controlling on pressure, we won't really have a "plateau pressure". I think that we basically want to take the inhale window and calculate the max pressure and the average pressure in that window - the max pressure is analogous to the Peak P reference above and the average pressure is the analogous to the plateau pressure (pplat) referenced above.

jkff, 2020-06-07:
Per offline conversation with @dtgerson this is actually 3 alarms:

1) "Max pressure" alarm - triggers if at any point pressure exceeds a specified threshold. This is a high-priority alarm, with both auditory and visual signal. It is latching, i.e. it persists until cleared, even when pressure has gone back under threshold. You can ACKNOWLEDGE it (suppressing the audio signal for 2 minutes) or RESET it (clearing both signals until it fires the next time).

2) "Max PIP overshoot" alarm - triggers if at any point pressure exceeds PIP + threshold (different threshold from 1). This is a medium-priority alarm with auditory and visual signal (different from the high-priority signals). It is also latching.

3) "Average PIP overshoot" alarm - triggers if the average pressure over the inhale phase of a breath exceeds PIP + threshold (maybe same threshold as 2?). Low-priority alarm, i.e. visual-only.

jkff, 2020-06-16:
Per offline discussion with @neelfirst and @ThomasSonsalla:
We'll only have one alarm here:

* Alarm condition is "Measured PIP of the most recent breath > threshold", where "measured PIP" is defined as "maximum patient pressure sensor reading over the course of the breath"
* Audio signal: beeps with the high priority tone (per ISO). Audio signal is latching - keeps beeping even if the condition is no longer active.
* Visual signals:
  * Measured PIP is highlighted in red, the pressure graph is highlighted in red. Visual signal is active whenever the alarm condition is active, i.e. it is non-latching.
  * An alarm banner appears. The alarm banner is active whenever the audio signal is active, shows the alarm text and has a "Pause" button".
* To make audio signal stop beeping, you press the "Pause" button. Then it stops beeping and the banner disappears (at least if there are no other active alarms).

Whether Pause means TIMED ACKNOWLEDGE or AUDIO PAUSE in terminology of ISO 60601-8, is still being discussed between the three of us.

jkff, 2020-06-18:
This is now completed after #551 .

closed 2020-06-18 by jkff