RespiraWorks/Ventilator #368: Implement "pressure not achieved" alarm

tapemaster, 2020-05-26:
#SR-22 Alarm shall trigger when inspiratory and PEEP pressure not achieved (i.e. disconnection)

dtgerson, 2020-05-30:
To add clarity, I think this DOES NOT refer to not quite hitting the pressure target and DOES refer to an inability to hit the pressure target.

If PEEP pressure cannot be maintained (e.g. inhale valve is opened as much as it can and the exhale valve is completely closed and the system still isn't at PEEP) - then this is a sign that something is wrong (e.g. a disconnection somewhere) and should trigger an alarm.

But if on one cycle, the system has a slightly low or slightly high PEEP, I think that is different (may be a separate alarm, but is less severe)

Similar with PIP.

jkff, 2020-06-18:
@dtgerson Is an alarm that detects "PIP < commanded value - threshold" good enough? That's what I'm making for now. Presumably it'd be just a matter of threshold. Or do we want to alarm on "PIP < commanded value over several cycles in a row"?

dtgerson, 2020-06-18:
@jkff 

Ideal would be two separate alarms.

Most important one, a disconnection alarm: "PIP < commanded value - threshold" at any point in time (this would use a large threshold and be a high priority alarm to signify something is very wrong because the pressure is off by a lot - ie disconnection). And I don't actually mean an instantaneous signal since noise may trigger it, I mean something averaged over a short period like ~100ms or something (but not across breaths)

Less important one (but still useful), a "low PIP" alarm: "PIP < commanded value - threshold" over several cycles in a row (lower threshold and to signify to the doctor that the ventilator isn't achieving the user settings and they might want to do something (e.g. change settings, maitenance, etc.))

Let me know if that makes sense

jkff, 2020-06-18:
For the disconnection alarm - measured PIP is updated once per breath, so it's gonna be several seconds before it's detected, but there isn't any room for a false positive due to noise on this one.

The second one - understood.

The alarm I added in #559 is more like the disconnection alarm, with a threshold of 2cm H2O.

What would be good thresholds for both of these alarms?

closed 2020-06-18 by jkff

dtgerson, 2020-06-18:
I assume you mean that it won't alarm until after the breath (but that if the single PIP value is below the threshold, it will alarm)? That's fine.

2cm H20 seems like a reasonable threshold to start for the second one - that seems like a bad threshold for the disconnection alarm (too low).

For disconnection alarm, I would say a decent number that scales with PIP would be (PIP-PEEP)/2. 10 seems reasonable for most cases. If you want a single value, let's go with 10 cm H20. We really should do some experiments of the blower at full power blowing through a fully open inhale valve, through the venturi, and then out into the environment (no patient tube) and measure what the pressure gets to - and then use that to guide our thinking. But I would expect that to be pretty small and so should probably be on the list of post covent

PR #559 (2020-06-18): Adds a "PIP not reached" alarm + some logic to combine it with "PIP exceeded"
Fixes #368 . 

Also fixes a bug that the Pressure Assist graph wasn't using the alarm.

Verified e.g. that if I tweak the commanded PIP here and there so that both alarms are active, then the alarm banner displays in red, and if I dismiss it then it's in yellow, if I dismiss that too then it disappears.