RespiraWorks/Ventilator #370: Design speaker solution for the alarms

tapemaster, 2020-05-26:
#SR-25 Alarm Volume shall be 60 to 80 dBA at one meter (+/- 5 dBA)

We need a design for the speaker that would conform to the requirement above

neelfirst, 2020-05-26:
The touchscreen we're using now has an integrated speaker, we just need to measure it with a sound meter at some point. @inceptionev

martukas, 2021-05-05:
@inceptionev measured this for Covent and posted video to Youtube. The speaker produced adequate dB

inceptionev, 2021-05-05:
This was measured on the prototype and documented to be 75dB at distance of 1m.  Video of the test is here: https://www.youtube.com/watch?v=_GUU8BG_yBc

This is good validation to proceed with this design, but obviously needs to be retested for any production design

closed 2021-05-05 by inceptionev