RespiraWorks/Ventilator #372: Solve flow measurement problem

tapemaster, 2020-05-26:
**What has to be done**
We need mathematically correct measurement of volume based on readings from the flow sensors.

**How do you know it has to be done**
Pressure sensors working alright at the moment. But to solve some of the challenges for ventilator, the volume and flow are also very important pieces of data. Inspiratory attempt is one of the things that depends on this.
https://respiraworks.slack.com/archives/C011ABVBS21/p1590516382457200

**Other**
There is work ongoing here: https://github.com/RespiraWorks/SystemDesign/issues/57

The whole slack channel `#flow-measurement`.

asmodai27, 2020-05-30:
Ran some simulations for volume computations through integration of flow, as well as some sensors characterization, and can outline several problems as of today:
- the flow computation carries the sign of pressure differential into the flow, this does not correspond to a physical phenomenon (a negative flow in the venturi still leads to a positive differential pressure).

- noise in the signals means that correcting the aforementioned error leads to a non-zero average flow for both inhale and exhale, and therefore to a  drift in volume. One way to solve this is implementing a deadband in the integrator to not take low flows into account. With my current sensor model, the deadband should be around 2.5 L/min, which is a little too high for the desired accuracy.

- there is significant sensor drift that leads to a changing bias in the measurements. The deadband solution takes care of some of this but how well depends on the actual drift, and is not an ideal solution, a better one would be to re-zero the sensors while the respirator is running, which is something Ethan is looking into.
A temperature sensor might help with correcting that drift, which looks like it is highly correlated with temperature.

tapemaster, 2020-06-25:
Difference in flow is calculated reliably enough.
What we need now is to have proper absolute flow measurement.

tapemaster, 2020-06-25:
The absolute flow measurement is extracted to a separate issue (#710). Closing this one now.

closed 2020-06-25 by tapemaster