RespiraWorks/Ventilator #376: Store hardware serial numbers in non-volatile storage.

neelfirst, 2020-05-26:
Please take the time to describe the issue in detail, so that other contributors can pick it up and implement or declare it done/obsolete in case if was completed out of band.

**What has to be done**
Demonstrate the recording of hardware serial numbers in EEPROM, OTP, or similar non-volatile storage.

**How do you know it has to be done**
Follow up from https://github.com/RespiraWorks/pcbreathe/issues/31. It is more expedient to serialize in software than hardware.

**Code pointers**
Give some thoughts on roughly where in code do you think the necessary changes might be located.

**Other**
Link the issue with a milestone, add appropriate labels, and mention dependent/blocked issues if any.

sglow, 2020-05-26:
This information can be stored in the flash memory of the STM32, or in the small OTP memory area as long as we definitely don't want to ever change it.

martukas, 2021-05-05:
@asmodai27 implemented non-volatile storage. There is memory reserved for unit serial, but it needs to be entered manually. Maybe this could be part of the deployment scripts for when the vent is run for the first time.

asmodai27, 2022-06-02:
We could run a sanity check in the debug scripts, which identify the serial number during initial communication with the controller, and ask the user to specify a value if the serial number is 0 (default value)
On a slightly different note, we may want to store alphanumerical values instead of simply a number.

PR #1302 (2022-08-11): Ventilator serial number saved as string
This changes the `0_ventilator_serial_number` debug variable from integer to string, so we can have semantically interesting unit descriptors in it, such as `v03e2` for the 2nd enclosed prototype of the v0.3 line. Will be even more important if various spin-offs and startups start using our software.

A knock-on effect is that the non-volatile memory implementation and tests had to be updated.

To make it work, I have also updated the debug interface, which did not have facilities for serializing string debug vars. This is the first writable one.

Close [... cut]
closed 2022-08-11 by martukas