RespiraWorks/Ventilator #393: Compiler optimizations in combination with StepperMotorInit are breaking the system

Miceuz, 2020-05-29:
**What is wrong**
Current master (951b144f61e7a5f421621b2e26e9d9bc6015898c) is broken in a weird way. 
 * `StepperMotorInit()` in `HalApi::init()` causes the watchdog reset in 250ms, system is caught in reset loop. Blinking LD2 is an indication for this. On a working system LD2 should shine constantly as it is connected to SPI SCK line that toggles at high frequency.
 * With `StepperMotorInit()` moved before watchdog initialization, the board does not reset, but there is no output produced from `dbgUart` or `rpiUart`.
* With `StepperMotorInit()` commented out system works as expected.
* With **-Og** build option added to build flags and the rest of the master branch left unchanged, system works as expected.

**How to reproduce**
 * `pio run -e stm32 -t upload` the firmware to the board. Observe the blinking LD2 on Nucleo board.
 * In `HalApi::init()` move `watchdog_init()` below `StepperMotorInit()` and flash the board. Observe LD2 is chining steady. Run `./utils/controller_debug.py`. Crashes with 
```Traceback (most recent call last):
  File "./utils/controller_debug.py", line 1118, in <module>
    cl.CmdLoop()
  File "./utils/controller_debug.py", line 92, in CmdLoop
    self.UpdatePrompt()
  File "./utils/controller_debug.py", line 85, in UpdatePrompt
    mode = SendCmd(OP_MODE)[0]
  File "./utils/controller_debug.py", line 916, in SendCmd
    raise Error("Invalid response, too short")
__main__.Error: Invalid response, too short
```
* Comment out `StepperMotorInit()` and reflash the board. Run `./utils/controller_debug.py`. Observe the command prompt is displayed. Enter `console`. Observe data from Cycle Controller coming in.
* Revert all changes to files and add `-Og` to `build_flags` in `platformio.ini`:
```
[env:stm32]
platform = ststm32
board = custom_stm32
build_flags =
  ${env.build_flags}
  -fstrict-volatile-bitfields
  -DBARE_STM32
  -mfpu=fpv4-sp-d16
  -mfloat-abi=hard
  -Wl,-Map,stm32.map
  -Wl,-u,vectors
  -Wl,-u,_init
  -Og
board_build.ldscript = boards/stm32_ldscript.ld
build_unflags = -std=gnu11 -std=gnu++14
extra_scripts = pre:boards/stm32_scripts.py
src_filter = ${env.src_filter} -<test/> -<src_test/>
```
Reflash the board and run `./utils/controller_debug.py` observe data is coming from Cycle Controller.

**How do you know it is wrong**
System should work providing lung ventilation for patient. 

**Code pointers**
Appearently something in `StepperMotorInit()` is optimized in a wrong way by the compiler. `-Og` build flag disables the most of the compiler optimizations unbreaking this.

closed 2020-05-29 by Miceuz

sglow, 2020-05-29:
I don't seem to be having this problem, even without a stepper board connected it works fine for me.
If you comment out the watchdog init, where does the code hang up in the stepper initialization?  Is it in SendCmd?  
```
while( cmd_ptr_ ){}
```
Seems like a likely spot to be hanging up, but `cmd_ptr_` is declared as volatile, so the compiler really shouldn't be optimizing out it's access to that.

Miceuz, 2020-05-29:
Yes, it's in SendCmd. If I add mark it non-optimizable for GCC:
```
StepMtrErr __attribute__((optimize("O0")))
StepMotor::SendCmd(uint8_t *cmd, uint32_t len){
[...]
```
It also compiles and works fine

Miceuz, 2020-05-29:
Ok, making cmd_ptr_ volatile pointer to volatile uint8_t fixes this `volatile uint8_t *volatile cmd_ptr_{0};`

sglow, 2020-05-29:
Excellent, thanks for tracking that down.  If you want to submit a PR with that change and removing the optimization setting I'll approve it immediately

commit 0048a4ca (2020-05-29): Declare cmd_ptr_ a volatile pointer to volatile data to avoid compiler optimizing out busy wait in StepMotor::SenCmd. Fixes #393
PR #395 (2020-05-29): Declare cmd_ptr_ a volatile pointer to volatile data 
…to avoid compiler optimizing out busy wait in StepMotor::SenCmd. Fixes #393
commit 116c1348 (2020-05-29): Declare cmd_ptr_ a volatile pointer to volatile data to avoid compiler optimizing out busy wait in StepMotor::SenCmd. Fixes #393