RespiraWorks/Ventilator #397: Get UX recommendations on display, user flows and alarms

tapemaster, 2020-05-29:
Beta milestone represents the final product without required documentation and certification.
This task is to think from beginning to the end about:
 - How the GUI of the ventilator should look like;
 - Review existing user flows;
 - Provide recommendations for alarms behaviour (visual and auditory).

Outcome of this task is a set of specifications which can be used by developers to improve the product.

jkff, 2020-06-18:
I think this is done, given the work @ThomasSonsalla has done on Slack (find on Slack by query e.g. "from:Thomas adobe")

martukas, 2021-01-18:
Some of the content of this decision should be published here in the ticket. Slack is opaque to the outside world.

martukas, 2021-05-05:
The work is done, but the underpinnings need to be documented properly with traceability