RespiraWorks/Ventilator #401: Better readme for /sample-data

martukas, 2020-05-29:
**What has to be done**
The readme should explain what the intent and the utility of this sample data is. The files have some inline docs explaining the exact nature of the experiment, but we should know how to replay them back into the system, or how to use them in manual tests of some sort?

**How do you know it has to be done**
Files were found while browsing the repo and it was not clear what to do with them. Made a README.md stub and owner of files was identified in PR:
https://github.com/RespiraWorks/VentilatorSoftware/pull/383

**Code pointers**
`sample-data/*`

martukas, 2020-05-29:
@jlebar tagging you here but not assigning you, no rush

jlebar, 2020-05-30:
> we should know how to replay them back into the system, or how to use them in manual tests of some sort?

No ability to do this at the moment.

martukas, 2021-05-05:
Actually the test data needs a new home, not in the repo.

martukas, 2021-05-12:
The `sample_data` is all plain text (uncompressed), and is now causing the software dir to take up over 300MB of space in the repo. The data does not deserve to be version-controlled. It is historical data, and will never be changed in any PR.

The sample or test data should be stored in a database. We should consider the following requirements:
* database should be publicly readable, easy to retrieve even not necessarily with any special credentials from our org
* data should be easy to link from the repo documentation pages
* data should be easy to clone to local machine
* data should be easy to upload (with proper credentials) by anyone running tests
* we should have automated scripts to help create the linked doc page and upload the data

martukas, 2022-04-21:
There is now a repository for the database that will be used to manage all future data:
https://github.com/RespiraWorks/test-data-browser

Python scripts are still needed in this repository as part of the controller debug interface, to automatically upload data to the database immediately upon performing the experiment. Some work was done in this direction on this branch:
https://github.com/RespiraWorks/Ventilator/tree/createDatabase
but may be outdated because it was focused on the "old format" data. Some of the mongo-related code might still be useful.

Proper implementation is dependent on implementation of an authentication solution described in:
https://github.com/RespiraWorks/test-data-browser/issues/3

asmodai27, 2022-06-02:
New home for test data is ready ([here](data.respira.works)), test automation is also ready (see #1166), missing piece of that puzzle is the automation of data upload.