RespiraWorks/Ventilator #403: Documentation for controller debug scripts

martukas, 2020-05-30:
**What has to be done**
Make a short guide on how `controller_debug.py` is to be used. What are prerequisites for running it? Does code have to be compiled a certain way? Are there any assumptions about hardware? Small sample of output on how it's useful.

**How do you know it has to be done**
This came up in a PR:
https://github.com/RespiraWorks/VentilatorSoftware/pull/383

**Code pointers**
Probably in `utils/README.md`, but there was talk of moving this script elsewhere, like a subdirectory?

**Other**
Probably will be done by @sglow or someone else that becomes equally familiar with this script.

sglow, 2020-05-30:
Addressed in PR #405

martukas, 2020-08-14:
This document got lost somewhere in the repo restructuring. I am briefly reopening this to return it to the repository proper.

closed 2020-08-14 by martukas