RespiraWorks/Ventilator #415: Controller must not assume a fixed air density value

jkff, 2020-05-31:
Per system requirements in Valispace (RW-SYS-063 "The device shall be have the ability to integrate a hot water humidifier...", RW-SYS-137 "Ventilator shall operate at altitudes below 10k ft"), the device has to operate under a variety of air conditions that affect its density.

**What is wrong**
Currently we [assume a fixed air density](https://github.com/RespiraWorks/VentilatorSoftware/blob/b7bc287d1cc417d11721858e0a376c20d76d7e27/controller/lib/core/sensors.cpp#L46)

Air density varies significantly [by altitude](https://www.engineeringtoolbox.com/air-altitude-density-volume-d_195.html) and [by humidity](https://www.engineeringtoolbox.com/density-air-d_680.html).

We should have some extra signals to compute the correct air density.

jkff, 2020-05-31:
CC @asmodai27

jkff, 2020-05-31:
Note: the current equation is not technically correct even at fixed altitude/humidity, because the pressure inside the ventilator is not atmospheric. It differs from atmospheric pressure only by a couple tens of cmH2O, which is a couple kPa which is a relative difference of a couple% in density.

martukas, 2020-07-23:
#814 and associated software would solve the air density part of this problem

inceptionev, 2020-07-23:
On the humidity side:
- Inhale side: doesn't matter, because humidifier is between the ventilator and patient.
- Exhale side: determine if humidity makes enough of a difference to violate requirements bounds, and implement fix if determined to be an issue.